Ultimaker/Cura #17157: Slicing Failed

AlexandraDorey, 2023-10-30:
### Cura Version

5.50

### Operating System

Windows 11

### Printer

Customer FF

### Name abnormal settings

yes fiddled with a lot

### Describe model location

reloading sometime works

### Describe your model

Models attached, have random problems with the cam-a & cam-b

### Add your .zip here ⬇️

The yarn weight seems to be the one that errors out the most consistently.

[bug report.zip](https://github.com/Ultimaker/Cura/files/13210288/bug.report.zip)
[image]
[image]

AlexandraDorey, 2023-10-30:
I assume you can see the profile when I saved my project file? if not let me know and I can upload them as well.

AlexandraDorey, 2023-10-30:
Aha. Seems to be related to support, turning supports on / off fixes it.

GregValiant, 2023-10-30:
Thanks for the report and the files.
I'm getting the same behavior you did.  If I make the supports "Normal" it will slice, but not with tree supports.  I didn't find any errors in the models.

I'll mark this as a duplicate slicing error and the Cura team will take a look.

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic