Ultimaker/Cura #17159: Cura 5.5.0 Update seems to have broken coasting

JoelAlgra, 2023-10-31:
### Cura Version

5.5.0

### Operating System

Windows 11 Pro

### Printer

IDEX modified ender 3V2 Running Marlin

### Reproduction steps

1) Downloaded this file to calibrate coasting [files.zip](https://github.com/Ultimaker/Cura/files/13211472/files.zip)
2) Enabled coasting in cura experimental mode
3) Set coasting to 0.08mm^3

### Actual results

Coasting works for skirt and smaller parts of model but for the main model there is no coasting except for on one layer (layer 7 or 8)

### Expected results

even coasting throughout every layer on all parts of the model 

### Add your .zip and screenshots here ⬇️

[CoastingIssue.zip](https://github.com/Ultimaker/Cura/files/13211504/CoastingIssue.zip)

GregValiant, 2023-10-31:
Thanks for the report.
Coasting works as designed...just not on the oval portion of that model.
This is with the coasting volume at 0.5 (to make the gaps easier to see).  No problem on my calibration shape but the oval is another story.  (I deleted the little pieces)
[image]

The only difference here is that I enabled "Remove all Holes".  Now the oval shows the coasting gap.
[image]

There is something about that thin walled oval that is causing Cura to ignore the Coasting setting while at the same time coasting shows up on the model next door.
I'll leave the bug label on this and the Cura team will take a look.

JoelAlgra, 2023-10-31:
Hey Greg, thanks for the work around, this will be useful for my calibration at any rate which will be nice. 

You saying thin walled made me think maybe if the walls were thicker it would fix it but I just got a different weird reaction.
(Again the coasting is scaled for easier viewing) 
[image]
Now the outer walls are coasting as expected but the inner walls don't want to be coasting. Maybe it has something to to with the respective walls being printed right next to each other with nothing (like infill even if set at 0%) printed in-between them?