Ultimaker/Cura #17160: Lässt sich nicht mehr slicen

Armin2577, 2023-10-31:
### Cura Version

5.5.0

### Operating System

Windows 11

### Printer

Elegoo Neptun 3 Max

### Name abnormal settings

Nein

### Describe model location

Nein

### Describe your model

Nein

### Add your .zip here ⬇️

[EN3MAX_1727-73-_77_bodyPart_d_Mid_FDM (repaired).zip](https://github.com/Ultimaker/Cura/files/13213084/EN3MAX_1727-73-_77_bodyPart_d_Mid_FDM.repaired.zip)

GregValiant, 2023-10-31:
Thanks for the report.
The project would not slice for me.
I had to rotate it and then change the Initial Layer Height to 0.23 and it sliced.
Those are normal workarounds for this slicing bug that started in 5..3 and although it has been worked on, it continues into 5.5.0.
[image]

I'll mark this as a duplicate and the Cura team will take a look.

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0