Ultimaker/Cura #17165: Ender 3 v3 SE should use dedicated nozzle clean area

0xDEAD, 2023-10-31:
### Is your feature request related to a problem?

Ender 3 v3 SE has a dedicated area on the left, (outside the print area) which should be used for Nozzle cleaning.
Instead the nozzle is cleaned on in the area X10.1/Y145.0 - X10.4/Y20.

I propose to update the cleaning gcode to use the dedicated area.

### Describe the solution you'd like

This will allow prints to use the full bed for printing and prevents objects being placed in the cleaning area.

### Describe alternatives you've considered

None.

### Affected users and/or printers

Users of Ender 3 v3 SE (support introduced in Cura 5.5)

### Additional information & file uploads

[image]

GregValiant, 2023-10-31:
The ability of the nozzle to move to a negative "X" location (or any negative location) is dependent on how the user has set their Home Offset location.  There may be a hard stop on moving to the left (or toward the front).
I have my purge lines moved to X0.1 going up and X1.1 coming back.  There are occasions when a skirt or model might need that space.  In those instances the prints are often large and the purge lines aren't needed so I comment them out in the StartUp.

mcuprojects, 2023-11-01:
I think this area on the print bed is meant not for purge lines but for scraping the nozzle during automatic Z-offset. If you perform auto-leveling routine you can see the nozzle being scraped there before probing Z-offset.

GregValiant, 2023-11-01:
The only time the nozzle should be at Z=0 is during the Auto-Home routine.  "Scraping" should never happen.  That's how build surfaces get ruined.  The way that M420 (or G29) is implemented in the firmware comes into that.  Some firmware will include an initial Z move so it doesn't happen.  It sounds like the firmware on that printer is not configured that way so it needs to be done in the StartUp.

This is from the StartUp gcode for your printer.  The G28 moves the nozzle to Z=0 and then ABL starts.
G28 ;Home
M420 S1; Enable mesh leveling
G92 E0 ;Reset Extruder
**G1 Z2.0 F3000 ;Move Z Axis up**
I would suggest you copy that first "Z" line and paste it in right after the G28 so the nozzle moves up before starting the ABL routine.  After the ABL runs it will move up again before doing anything else.
Whenever the Z is at 0, the nozzle must always move up before moving in the X/Y.  I would also suggest you don't have moves to negative locations on any axis.

M220 S100 ;Reset Feedrate
M221 S100 ;Reset Flowrate
G28 ;Home
**G1 Z2.0 F3000 ;Move Z Axis up**
M420 S1; Enable mesh leveling
G92 E0 ;Reset Extruder
**G1 Z10.0 F600 ;Move Z Axis up**
M109 S[material_print_temperature_layer_0]
G1 **X0.1** Y20 Z0.28 F5000.0 ;Move to start position
G1 **X0.1** Y145.0 Z0.28 F1500.0 E15 ;Draw the first line
G1 **X2.1** Y145.0 Z0.28 F5000.0 ;Move to side a little
G1 **X2.1** Y20 Z0.28 F1500.0 E30 ;Draw the second line
G92 E0 ;Reset Extruder
G1 E-1 F1800 ;Retract a bit
G1 Z2.0 F3000 ;Move Z Axis up
G1 E0 F1800

The build plate is 235 x 235.  Within the definition file - the "printable area" is defined as 220 x 220.  That makes for a very conservative 7.5mm "safety area" all around.  It should not be a problem for a user to increase that to 230 x 230, but then the print head might be right up against the end stop switch when the Home Offset is set in the printer.  If that is true, then there can't be a negative move because the print head is up against a hard stop.

PR #17166 (2024-03-29): Move nozzle clean to designated area
# Description

Move nozzle clean movements to designated area of the heatbed.

This fixes #17165 

## Type of change

- [X] Printer definition file(s)

# How Has This Been Tested?

- [X] Sliced and tested gcode on a Ender 3 v3 SE
Host system: Windows 10

- [ ] My code follows the style guidelines of this project as described in [UltiMaker Meta](https://github.com/Ultimaker/Meta) and [Cura QML best practices](https://github.com/Ultimaker/Cura/wiki/QML-Best-Practices)
- [X] I have read the [Contribution guide](https://github.com/Ultimaker/Cura/blob/main/CONTR [... cut]