Ultimaker/Cura #17171: "unsupported operand type(s) for /: 'NoneType' and 'int'"

sumone-lol, 2023-11-01:
### Cura Version

5.5.0

### Operating System

Windows 11

### Printer

Creality Ender 3 Pro

### Name abnormal settings

N/A

### Describe model location

Location and rotation doesn't matter

### Describe your model

Haven't sliced it before but I think it should still be fine, it's a pretty simple model

### Add your .zip here ⬇️

Model I tried slicing (I'm not very experienced don't judge me) [USB Case.zip](https://github.com/Ultimaker/Cura/files/13222129/USB.Case.zip)

Error message I received:
[image]

This only started happening after I updated to 5.5.0, it was fine in 5.4.0

GregValiant, 2023-11-01:
With the model loaded use the "Help | Show Configuration Folder" command.
Within the Configuration Folder will be a file "Cura.log".  Right click on it and select "Send To... Compressed (ziped) folder" and post the zip folder here.

[image]

sumone-lol, 2023-11-01:
[cura.zip](https://github.com/Ultimaker/Cura/files/13222376/cura.zip)

GregValiant, 2023-11-01:
Thanks.
@MariMakes or another member of the Cura team will take a look.

yzheka, 2023-11-06:
I have noticed that Arc_Welder plugin causing this issue. Disabling it solves the problem

GregValiant, 2023-11-06:
@fieldOfView can you look at this one?

sumone-lol, 2023-11-06:
Huh, strange. I went to test it and turn off arc welder which worked. But then after I re-enabled it, it still worked. Not sure what went on there :L  I'll probably reopen this if it happens again in the future.

closed 2023-11-06 by sumone-lol