Ultimaker/Cura #17173: Slicing fail when modify horizontal expansion.

jaetma, 2023-11-01:
### Cura Version

5.5.0

### Operating System

Windows 11

### Printer

Geeetech A20M

### Name abnormal settings

horizontal expansión with negative value (-1, -.8)

### Describe model location

After reset all model transforms it slices

### Describe your model

Yes i have sliced this kinds of models before

### Add your .zip here ⬇️

[GA_base-bone-4922588354.zip](https://github.com/Ultimaker/Cura/files/13223202/GA_base-bone-4922588354.zip)

GregValiant, 2023-11-01:
Thanks for the report.
You have Horizontal Expansion set to -.08 for E2 but E1 is set to 0.0.  With the HE set the same for both extruders it slices fine.

It doesn't mention in the tooltip that they have to be the same, but it appears that they do in fact have to be the same.
There is also some odd results in the preview because the line width is different for each extruder and the flow is a lot different for each extruder.
The Cura team will take a look.

jaetma, 2023-11-01:
Yes, that is on pourpose, i only got this error since 5.5.0 update (coming from 5.2)