Ultimaker/Cura #17176: Setting to put z seam on support if possible

Garbg, 2023-11-01:
### Is your feature request related to a problem?

There is no setting to prefer putting z seams on supports.

### Describe the solution you'd like

There should be a setting in the z-seam positioning settings to put the z seams on supports as much as possible.

### Describe alternatives you've considered

There are no alternatives.

### Affected users and/or printers

This will help people who struggle with very visible z seams and blobs on their printer. This would help because if the z seam is on the support the defect wouldn't mater because you would just get rid of the support.

GregValiant, 2023-11-01:
Sorry but the Z seam is a function of the start and end of the outer wall extrusion.  Each separate support structure already has a Z-seam.  The model has a Z-seam on every "island" of the print.  It is the nature of FDM.  The only thing that can possibly eliminate it is to never stop extruding which is what Spiralize does.  An alternative is to use a resin printer.

I'll go ahead and close this as it is impossible.  I hope you understand.

closed 2023-11-01 by GregValiant

Garbg, 2023-11-02:
I meant the layer change, sorry

GregValiant, 2023-11-03:
If there are supports on a layer they always print first.  That puts the layer change at the supports every time.  The nozzle moves from the model, up a layer height, and does the supports.
The alternative would be to print the supports last.  In that case it's simply the reverse of what is already there...move up and then from support to the model at layer change.
The Z-seam is because there is a stop on the outer wall.  There will always be a stop on the outer wall.  That's how FDM works.