Ultimaker/Cura #17178: [5.5.0] [5.4.0] Support Interface Extruder creates Brim on First layer around normal Brim, even when "Enable Support Brim" is deactivated.

SonGoku10c, 2023-11-02:
### Cura Version

5.5.0

### Operating System

Windows 10

### Printer

Custom CoreXY

### Reproduction steps

-Change Support Interface Extruder to a different Material (VTOOL1 for me)
-Enable Support Interface and both Roof and Floor
-Deactivate Enable Support Brim
-Enable Brim at Build Plate Adhesion with your normal Material (VTOOL2 for me)

### Actual results

There is a Brim with the Support Interface Extruder material around the normal Brim.

### Expected results

No Support Interface Extruder Material Brim around the normal one. ^^

### Add your .zip and screenshots here ⬇️

[SupportInterface.zip](https://github.com/Ultimaker/Cura/files/13232774/SupportInterface.zip)
[image]
[image]
[image]
[image]
[image]

GregValiant, 2023-11-02:
Thanks for the report.
You will get the same effect with a skirt as well.  Part of the function of the skirt/brim is to get the nozzle flowing nicely.  If both extruders will be used in the project then they both will be used in the brim or skirt.  That is expected behavior rather than a bug.
Is it causing a problem?  It's a skirt/brim so it will be discarded.
I'll leave the bug label on this and leave it open for a bit so you can respond.

SonGoku10c, 2023-11-02:
Hi.
The problem is, that I only use that material in 2 other layers. And when I change the material I flush it out of the hotend with the new material and because of that additional brim in the first layer, I waste extra material just for something I don't need there. I use a Y-connector above the Hotend for the material switch. Here is the Extruder start code I use for every VTool:
`; virtual Extruder 2 Start Code
G60 S0 
G91
M83
G1 E-10 F3600
G1 Z10 F3600
G90
G1 X100 F6000
G1 Y0 F6000
G92 E0
G1 E300 F300
G1 R1 X0 Y0 Z10 F6000
; END virtual Extruder 2 Start Code`

And this is the End Code:
`; virtual Extruder 2 End Code
M83
G1 E-150 F3600
G90
; END virtual Extruder 2 End Code`