Ultimaker/Cura #17179: Help me please >.>

TheNinjineer123, 2023-11-02:
### Cura Version

5.5.0

### Operating System

Mac OS

### Printer

Ultimaker S5

### Reproduction steps

I am printing a mask, if that matters, out of Black Tough PLA and PVA as supports. I have double checked all of the settings and the print got a full day and some change into it before I went to work today. When I came back home, I was greeted with a stopped printer and a blank unresponsive screen. The print did not finish and moreover, I can even see where the print abruptly stopped and dragged some Tough PLA over the print in progress. Without the screen I have no idea what happened and because I am new to 3D printing (at least new enough to not be familiar with this issue since just about every other print was fantastic) I have no clue if the print can be resumed or what in the world to do. I am also not sure if I am even in the right forum, so direction and guidance on any of these fronts would be appreciated and thank you in advance.

### Actual results

Without the screen I have no idea what happened and because I am new to 3D printing (at least new enough to not be familiar with this issue since just about every other print was fantastic) I have no clue if the print can be resumed or what in the world to do. I am also not sure if I am even in the right forum, so direction and guidance on any of these fronts would be appreciated and thank you in advance.

### Expected results

I would have loved a responsive screen and a finished print.

### Add your .zip and screenshots here ⬇️

:( I don't know if I can continue the print and i cant see the screen, so I want to find out if I have to dump out of the print or just resume it.

GregValiant, 2023-11-02:
Your report here is clear enough to get some help.  Not really a bug report though.  Go over the the [Ultimaker Forum](https://community.ultimaker.com/forum/106-ultimaker-3d-printers/) and post this.  gr5 is a moderator there and if you put "@gr5" into your post he will be notified and should respond in a timely fashion.

In the meantime I'll leave this open for a bit as I'm not an UltiMaker guy but one of the team members may see it here and respond.  I'll change it from a bug to a discussion though as it is unlikely to be a Cura problem.

gr5, 2023-11-02:
There's a log file even if screen is blank.  Go post over on the forums for more help.

GregValiant, 2026-09-10:
I'll go ahead and close this.

closed 2026-09-10 by GregValiant