Ultimaker/Cura #17183: cura

Aarenwain, 2023-11-02:
### Cura Version

5.4.0

### Operating System

windows 10

### Printer

anycubic kobra go

### Reproduction steps

will not slice in cura but will with others
[AKGO_TORSO.zip](https://github.com/Ultimaker/Cura/files/13239169/AKGO_TORSO.zip)

### Actual results

nothing just says error report bug

### Expected results

nothing i just need it to slice with supports

### Add your .zip and screenshots here ⬇️

[Uploading AKGO_TORSO.zip…]()

GregValiant, 2023-11-04:
Thanks for the report and the file.
There is a bug in 5.4.0 that can keep models from slicing when they are at the midpoint of the Cura build plate.  There are 3 workarounds and I needed to do all three to get this to slice.

1. Move the model on the Cura build plate.
2. Rotate the model in Cura.
3. Make a small change to the Initial Layer Height.

The bug was worked on for 5.5 but there are still reports of this behavior.  I'll mark this as a duplicate and close this as the Cura team is aware of this bug.
I rotated and moved the model and changed the Initial Layer Height to 0.33 and it sliced.
[image]

closed 2023-11-04 by GregValiant

Aarenwain, 2023-11-04:
Thank you very much

[image]<https://user-images.githubusercontent.com/64202104/280466040-849c8538-457d-4ca0-8109-67b282fc60f0.png>