Ultimaker/Cura #17185: Slicing Failed

PDignam, 2023-11-02:
### Cura Version

5.5.0

### Operating System

Windows 11

### Printer

Anycubic Vyper

### Name abnormal settings

Slicing Tolerance Exclusive

### Describe model location

center flat

### Describe your model

A round fidget Spinner

### Add your .zip here ⬇️

[AV fidgit_planetary Weedo - Blue PLA lw 0.4mm lh 0.2mm if 20% ext1 190.0C bed 68C.zip](https://github.com/Ultimaker/Cura/files/13241806/AV.fidgit_planetary.Weedo.-.Blue.PLA.lw.0.4mm.lh.0.2mm.if.20.ext1.190.0C.bed.68C.zip)

GregValiant, 2023-11-04:
Thanks for the report.
This would not slice for me in "Exclusive" either.  The Cura team will take a look.

The same sort of model was here in another bug report.  Slicing with "Middle" and the Outer Wall Inset at .05 worked and I'm sitting here playing the one I printed.

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0