Ultimaker/Cura #17199: Cura ignoring initial layer speed setting when set to skirt, 3 lines

zebrabunny, 2023-11-03:
### Cura Version

5.5, 5.4

### Operating System

Win11 22H2 

### Printer

Anycubic Kobra Max (early version)

### Reproduction steps

Fresh install of 5.5 after new hard drive install.
Printed a file that I have printed before
First layer, skirt prints at proper speed, but rest of the layer fails to run at the 20mm/sec first layer speed.
The rest of the print (80mm/sec except for outer walls) prints OK
I backed down to 5.4 which I had been using for some but the problem remained with the same project file, enclosed.

dolphin@strangeanimals.net   <- email for questions

### Actual results

print suffers from bad adhesion on small details in first layer... slowing the printer manually makes it better. (I am running a modded version of wabbit-ware on my anycubic but there have been no problems until now)

### Expected results

Normal print speed control

### Add your .zip and screenshots here ⬇️

[AKMAX_Baroque_Frame_Fixed_Mesh_Flat_base.zip](https://github.com/Ultimaker/Cura/files/13249289/AKMAX_Baroque_Frame_Fixed_Mesh_Flat_base.zip)

zebrabunny, 2023-11-03:
After watching the print carefully, at some point during the first layer the print does slow down (after a couple passes around the baroque shapes of the file) but before that first layer is finished.

Austenite2, 2023-11-13:
Could this be the same as issue 17210 ?

zebrabunny, 2023-11-14:
It does look it could be, but the problem persisted in cura 5.4 when I
tried downgrading… checking the speed color graph, indeed it seems to be
the same problem. (I had set the skirt speed high because I had been
manually tweaking the printer speed down during the first part of the
print, ignore that.

[image: image.png]

zebrabunny, 2023-11-14:
However the workaround reported by another poster, of setting Print Thin
Walls and setting Minimum Feature Size to 0 - merely resulted in a slicing
failure. :(

zebrabunny, 2023-11-14:
[image]
here's my speed/color graph

zebrabunny, 2023-11-28:
I have confirmed that cura 5.6b2 still has the bug. (and 5.4, 5.5)
however cura 5.3.1 works properly - both using the default kobra max settings
and opening the project that I supplied, with the modified parameters.

GregValiant, 2026-10-25:
I'll go ahead and close this as stale.

closed 2026-10-25 by GregValiant