Ultimaker/Cura #17206: Fuzzy Skin Consistency

tedbergstrand, 2023-11-03:
### Cura Version

5.5 beta

### Operating System

Windows 11

### Printer

Creality Ender 3 Pro

### Reproduction steps

1.) Make model with curvature on the Z axis and holes starting midway through curve. 
2.) Turn on Fuzzy Skin in Cura
3.) Preview

### Actual results

Fuzzy skin skips a few layers where the holes start. My guess is that it's doing the interior hole geometry first and then forgets to do the fuzzy skin when it does its outer layer pass.

### Expected results

Fuzzy skin should still be applied on these layers.

### Add your .zip and screenshots here ⬇️

This is in the 5.5 beta and is consistent for me across several models.

[fuzzy-skin-problem.zip](https://github.com/Ultimaker/Cura/files/13252791/fuzzy-skin-problem.zip)
[image]

Tad009, 2023-12-10:
Same issue here. 

I would also note that it only does this when I have "Fuzzy Skin Outside Only" Checked. If it is the whole thing then it works fine.
[image]

GregValiant, 2026-10-25:
I'll go ahead and close this as stale.

closed 2026-10-25 by GregValiant