Ultimaker/Cura #17210: Cura 5.5 first layer perimeter speed inconsistent

Mrjimbo25, 2023-11-04:
### Cura Version

5.5 stable

### Operating System

Windows 11

### Printer

Custom i3 - klipper

### Reproduction steps

Slide example file with multiple perimeters on first layer (e.g. Voron door hinge for switchwire) and the perimeter speed is not consistent to either the "initial layer speed" or "top/bottom speed" - it is observed for one of the perimeters but the second is printed at normal perimeter speed, with the correct slower speed for the infill/bottom layer. This happens again on the 2nd and 3rd layers until the layers are at normal speed (I have the first 3 layers slower)

always replicated on the smallest perimeter, repeatable when printing multiple items, all the smaller perimeters are at full speed

I have tried uninstalling, deleting appdata/roaming/cura 5.5 before reinstalling and issue persists.

### Actual results

Initial layer perimeters print too quickly and ignore first layer / slow down settings

### Expected results

consistent first layer perimeter speeds to desired settings.

### Add your .zip and screenshots here ⬇️

[image]
[image]
[image]

thejudge156, 2023-11-11:
I am having this issue as well. Using Cura 5.5, would really appreciate a fix.

Austenite2, 2023-11-13:
I am having similar issue, Cura 5.5.

I think it might be related to _Initial Layer Horizontal Expansion_ being accidentally set to a postive value, and in my case 0.28mm or larger.  0.27 does not trigger, nor does any negative value.

0.6mm nozzle, 0.3 layer height, 0.25 initial layer height, 0.72 line width.

DanielGerlinghoff, 2023-11-13:
I am having the same issue as in your last picture, where the regular print speed is applied to some walls in the initial layer.

My workaround is to check `Print Thin Walls` and set `Minimum Feature Size` to 0. But depending on your model, this might have other negative consequences.

Kathode-Ray, 2023-11-14:
I'm having the same problem, but only with some particular STL's. The above workaround works for me as well.

Cura seems to ignore the initial layer speed in some cases, here's another example:

[image]
[image]

Mrjimbo25, 2023-11-15:
Have tried the solutions above but still face the same issue on mine!

pedrolamas, 2023-11-20:
I'm experiencing this same issue on Cura 5.5...

You can see below the first layer of the same model, sliced in Cura 5.2 and 5.5

## Cura 5.2

[image]

## Cura 5.5

[image]

pedrolamas, 2023-11-20:
Another test, Cura 5.5 using "Standard" Ender-3 profile, just set the speed to 100mm/s:

[image]

This should all be blue (slow), but it is not!

pedrolamas, 2023-11-20:
Fresh install of Cura 5.6 beta 2, stock Ender-3 profile with 100mm/s speed, also shows the same problem!

[image]

Kathode-Ray, 2023-11-20:
I can verify that. The initial layer speed is still ignored, but the skirt is o.k. in 5.6.0-b2.

MariMakes, 2023-11-21:
Hi all,

Welcome to the UltiMaker Cura Github 🚀 
Sorry, it took us a while to get back to you 😞 

I'm seeing a few different issues happening here. 

Indeed a number of them seem to be related to a slow down related to [Minimum Feature Size](https://github.com/Ghostkeeper/SettingsGuide/blob/master/resources/articles/shell/min_feature_size.md) if you don't want that to happen, you can reduce the setting to 0 to disable it. 

The issue from @pedrolamas seems to be related to settings introduced to enable variable line width in the Arachne Engine.
You can read more about it here: https://support.ultimaker.com/hc/en-us/articles/4792077687068

If you set your Flow Equalization Ratio to 0, you won't see any speed changes:
[image]

@Mrjimbo25, I'm especially interested in the hinge-like models that you shared. That behavior doesn't look as I expected
[image]

Do you have a project file for us? It contains the printer and settings we need for troubleshooting. 
To save a project file go to File -> Save project. Please make sure to .zip your project file. For big files, you may need to use WeTransfer or similar file-sharing sites.

You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser.

pedrolamas, 2023-11-21:
Thank you for your reply @MariMakes, however I don't think that is the problem.

I've just done a fresh install of Cura 5.6 beta 2, added a brand new Ender-3 printer profile, set the speed to 100mm/s and (following on your comments) flow equalization ratio to 0:

[image]

I can see clearly there the difference in speeds!

Here is the project file for your reference:

[CE3E3V2_Gingerbread_Kit_A.zip](https://github.com/Ultimaker/Cura/files/13424958/CE3E3V2_Gingerbread_Kit_A.zip)

Mrjimbo25, 2023-11-21:
HI @MariMakes, thanks for taking a look into this. 

Please find attached the same file (organized slightly differently on the build plate) but exhibiting the same speed difference on the smaller perimeter of the hinges. Sliced with my profile on 5.5.0

[
[PI3MK3M door_hinge_x4 Hatchbox ABS lw 0.4mm lh 0.2mm if 15% ext1 235C bed 100C.zip](https://github.com/Ultimaker/Cura/files/13429400/PI3MK3M.door_hinge_x4.Hatchbox.ABS.lw.0.4mm.lh.0.2mm.if.15.ext1.235C.bed.100C.zip)
](url)

pedrolamas, 2023-11-28:
I just tried the recently released Cura 5.6, this issue is still there but there is no mention of it on the 5.6 release notes?

[image]

Those initial layer walls should all be printed at 20mm/s, but they are more than that!

[CE3E3V2_Gingerbread_Kit_A.zip](https://github.com/Ultimaker/Cura/files/13490370/CE3E3V2_Gingerbread_Kit_A.zip)

Mrjimbo25, 2023-12-06:
Good morning all, not to chase this issue, but some more information, I decided to do a full OS reinstall recently due to some other issues, and installed a fresh copy of 5.6 from Ultimaker's website, however the issue still persists on my system. Happy to share a project file again if it helps when I get home

zeisig123, 2023-12-06:
I have the same problem. I tried with Cura 5.5.0 and 5.6.0, Thin Walls enabled (with values 0 and 0.05) and Flow Equalization Ratio = 0.
However, I noticed that the walls are printed at the speeds "Top Surface Outer Wall Speed" and "Top Surface Inner Wall Speed".

The problem occurs with very flat objects (in my case 2 layers). Then “initial layers” overlap the “top surface skin layers”. 
**So he start immediately to print the “Top Surface Skin Layers”. _But that only affects the walls._**
My solution: I set the "Top Surface Skin Layers" to 0 and now flat objects work too.

Settings (wrong wall speed): Thin Walls enabled (=0) / Flow Equalization Ratio = 0 / Top Surface Skin Layers =2

[image]

Settings (correct wall speed): Thin Walls enabled (=0) / Flow Equalization Ratio = 0 / **Top Surface Skin Layers = 0**

[image]

So the problem is that when slicing the “initial layers” the wall speed is overwritten by the wall speeds of “Top Surface Outer Wall Speed” and “Top Surface Inner Wall Speed”.

DuckersMcQuack, 2023-12-15:
Same issue here. On 5.5 as well as 5.6. Rolling back to 5.4 or earliest which doesn't have this issue.

[image]

TerranEarth, 2024-01-11:
I have a similar problem on 5.5 and 5.6.
Enabling thin wall printing normalizes the speed of the outer perimeters.
Disabling causes the walls of the first layer to print at speed, ignoring the modifier of the first layer.

step4, 2024-01-22:
Same Issue appears for me as well.
Also using the post processing script `ChangeAtZ` won't override the print speed for the first layer.

All other settings like `Thin Walls enabled/ Flow Equalization Ratio = 0 / Top Surface Skin Layers = 0 ` are not working for me.

Flight, 2024-02-11:
Cura 5.6.0. Same issue, setting `Minimum Feature Size` to 0.0 or 0.05 helps. Spent 2 days trying to understand what is wrong with my adhesion until realised that I'm printing the first layer at 150 mm/s 😢

pedrolamas, 2024-02-12:
Out of curiosity, I tried the Cura 5.7.0-alpha.0 build, added a new Ender-3 printer profile, set the speed to 150mm/s (no other changes), sliced a file and this is what I got:

[image]

The whole initial layer should be printed at 20mm/s, which is correct!

It seems we are finally getting a fix for this problem?

KJ7LNW, 2024-02-16:
> I can verify that. The initial layer speed is still ignored, but the skirt is o.k. in 5.6.0-b2.

We had the same issue and setting minimum feature size to 0.01 works but 0.05 is too big.   I've attached our model in case it is useful for troubleshooting.

Is there a way to tell where on the model that there is a <0.05mm feature so we can fix our design?

Here is the .3mf renamed to .zip:
[motor triangle.3mf.zip](https://github.com/Ultimaker/Cura/files/14316380/motor.triangle.3mf.zip)

Kathode-Ray, 2024-02-21:
> Out of curiosity, I tried the Cura 5.7.0-alpha.0 build, added a new Ender-3 printer profile, set the speed to 150mm/s (no other changes), sliced a file and this is what I got:
>
> The whole initial [...]

Same here, it seems that 5.7a0 fixes the initial layer speed problem.

[image]

Here's the model that I used, for those who are interested. This doesn't slice correctly in 5.6:

[Heart_Key_Ring.zip](https://github.com/Ultimaker/Cura/files/14357859/Heart_Key_Ring.zip)

Experte9, 2024-06-30:
Is this fixed in 5.7.2?

Austenite2, 2024-06-30:
> Is this fixed in 5.7.2?

Since my comment back in November 2023 in Cura 5.50, I have since been using 5.6 and currently 5.7.0, and I have not experienced the issue again.  Just tested in 5.7.0 and set initial layer horizontal expansion to a large positive value (+0.5) and did not reproduce the issue.  

So my particular case seems to be fixed, although I note others proposed different causes/fixes.