Ultimaker/Cura #17215: TPU 95A matyerial temperature settings not transferring to settings

Wolfhowell, 2023-11-04:
### Cura Version

5.5

### Operating System

Windows 10

### Printer

Elegoo Neptune 3 Pro

### Reproduction steps

1. Duplicate the Generic TPU 95A material from materials manager
2. Change the temperature settings for the material
3. Activate the profile
4. Note the temperature settings will be set to 230. 
5. Remove the "A" from the end of the Material Type description, leaving TPU 95
6. Magically, the temperature will now reflect the materials setting.

### Actual results

By omitting the "A" from TPU 95A in the Materials Type descriptor in the Manage Materials window, you are able to set the temperature required. Put the "A" back on the end of "TPU 95" and you're doomed to print at 230 unless you manually override. 

### Expected results

I expect to be able to name it TPU 95A and it print at the temperature I set.

### Add your .zip and screenshots here ⬇️

N/A

GregValiant, 2023-11-04:
Thanks for the report.
The Cura team will take a look.  I followed your steps and the temperature was 228.  When I went back and made a second adjustment it correctly changed to 230.  It may be a problem with the setting box in "Manage Materials".
Do you happen to have the "Material Settings" plugin loaded?

Wolfhowell, 2023-11-04:
A user on Reddit pointed me to this plugin as well. I have installed the 3.6.2-DEV version of the plugin (it was not previously installed) and have replaced the "Default Printing Temperature" with "Printing Temperature" and ":Default Build Plate Temperature" with "Build Plate Temperature." This seems to have the desired effect.  Is this working as designed, or an issue with how the Neptune 3 printer profile handles materials settings?

GregValiant, 2023-11-04:
When the dialog boxes were re-worked for 5.0 there were problems here and there.  Some of the problems have lingered.  The problem with the Material Settings plugin showed up in 5.5 because of an internal change in Cura.  The problem seems to be resolved by the author of the plugin.

Can we go ahead and close this?

Wolfhowell, 2023-11-04:
Yes, with the plugin it seems to be working as expected. We can close. Thank you.

closed 2023-11-04 by GregValiant