Ultimaker/Cura #17220: not slicing or taking ages to slice on high end pc

Ken-1991, 2023-11-04:
### Cura Version

5.5.0

### Operating System

windows 10

### Printer

ender neo 3

### Describe model location

flat

### Describe your model

not sliced successfully tried mesh fix

### Add your .zip here ⬇️

[kens file.zip](https://github.com/Ultimaker/Cura/files/13257056/kens.file.zip)

GregValiant, 2023-11-04:
Thanks for the report.
I've seen this slow slicing on models like this that have a lot of features within 1 or 2 layers.  Buildings with rough hewn wood floors, or rough tile floors, lithophanes, that sort of thing.

The Cura team will take a look.  I sliced it in 5.2.2 and it finished, but it took a long time.
Mesh fix has some rudimentary ability to fix models.  The errors in your models are well beyond it's capabilities.
Here is a report on the models.  But even after repairing them they wouldn't slice in 5.5.0.
-> Analysed your file:
--> 220 Naked edges (?)
--> 64 Planar holes (?)
--> 0 Non-planar holes (?)
--> 82 Non-manifold edges (?)
--> 133 Inverted faces (?)
--> 0 Degenerate faces (?)
--> 7 Duplicate faces (?)
--> 0 Disjoint shells (?)
-> Repairing: 100.00%
----- Repair completed in 42361ms ------
-> Vertex count changed from 150049 to 150260 (+211)
-> Triangle count changed from 300020 to 301374 (+1354)

[image]

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic