Ultimaker/Cura #17224: Shop

Gadfrod, 2023-11-04:
### Cura Version

5.4.0

### Operating System

Windows 10

### Printer

Creality Ender 3

### Describe your model

Is a Shop

### Add your .zip here ⬇️

[CE3E3V2_Local Comercial 3D Nuevo.zip](https://github.com/Ultimaker/Cura/files/13257788/CE3E3V2_Local.Comercial.3D.Nuevo.zip)

GregValiant, 2023-11-05:
Thanks for the report.  The model is error free and I can't get it to slice in either 5.4 or 5.5.  5.2.2 sliced it fine.

The Cura team will take a look.
@MariMakes this is a very simple non-organic model.

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic