Ultimaker/Cura #17225: 0.8mm nozzle and Adaptive Layers

darkjacky, 2023-11-04:
### Cura Version

5.5.0

### Operating System

Windows 11

### Printer

Creality Ender 3 V2 Neo

### Reproduction steps

First I installed a 0.8mm nozzle as it prints faster. Then in Cura I did what I usually do which is adaptive layers for the  added quality while reducing the print time.

### Actual results

The thing that was printed had a bulge at the exact layers where the layer height changed.

### Expected results

I would expect Cura to be able to correctly offset the layers so that doesn't happen. Which is my assumption of what is not happening.

### Add your .zip and screenshots here ⬇️

[image]
[slide.zip](https://github.com/Ultimaker/Cura/files/13257811/slide.zip)

GregValiant, 2023-11-05:
Thanks for the report and the file.
I read the gcode into AutoCad.  The sides and back of the model are exactly straight with no bulges.

Adaptive Layers varies the layer height but not the line width.  Cura correctly adjusts the XY to the wall of the model regardless of the layer height and so the AutoCad representation of the model is exactly as I would expect.  With that particular model, since it's a single wall print, there is also no requirement to adjust the line width.

Since that model doesn't have any topography changes (it's a non-organic shape) it doesn't require Adaptive Layers.  I'm surprised that the layer height varied at all.  I would have expected that it would go to the max layer height and stay there.

ABS can react oddly to the amount of heat put into a model.  That's why it is recommended to be printed in a heated build volume.  Since the layer height is changing, the volume of hot material being added per layer varies.  You have the layer cooling fan at 60% and ABS is often printed with no cooling fan to avoid warpage.  I wonder if that behavior you are getting is a function of cooling.

I really don't see that this is a problem with the gcode that Cura generated.  How does the model print with Adaptive Layers turned off and how does it print with an easier material like PLA?

What firmware are you running on the printer?  The startup gcode contains non-Cura replacement words and some other lines I was surprised to see.

darkjacky, 2023-11-05:
I used PETG for that. After printing and finding it like this I went into Blender to remove some vertices to straighten the model. That removed the extra layer height changes near the top so fixed the print. It probably prints fine without adaptive layers however there seems to be a tiny gap in the model that causes a hole near the bottom that just doesn't want to fix itself so I was hoping that using adaptive layers would reduce the hole size.

I added some code to do some custom magic to speed up the heating process. Like heating the bed to the target -15 degrees then continue heating to the target while also heating the nozzle. Also after the print it retracts the filament so before the start it needs to push it back some. (If only I wrote the lines part correctly though. Doing E30 and then E15 is not my smartest move.) 60% fan speed is from testing towers that gave the best result in terms of stringing and strength. Well technically 215c gave the best result but I need some strength on this part so its printed hotter.

No idea how it would print with PLA as I only have colorless PLA and I need this to be white to match the color of the thing I put this into.

Then the firmware, Marlin 2.0.8.3 from the Creality website. I tried using another firmware but that one does not have the correct display update (TJC_SET) so it does not run correctly.

I wonder if the bulge is 0.4mm sadly I can't check unless I print another one as I gave it to my sister. Because that would indicate there is a slicer problem. Though you say there isn't which is why this is so weird. I also didn't see it when I checked the preview. It also always prints perfectly fine with adaptive layers just not this time.