Ultimaker/Cura #17229: Slicing Fails to finish

SlingerMD, 2023-11-05:
### Cura Version

5.5.0

### Operating System

Windows 11

### Printer

ender 3

### Name abnormal settings

no

### Describe model location

all positions

### Describe your model

soap dish from Thingiverse have printed previously

### Add your .zip here ⬇️

[CFFFP_Full size soap dish.zip](https://github.com/Ultimaker/Cura/files/13258720/CFFFP_Full.size.soap.dish.zip)

SlingerMD, 2023-11-05:
just updated

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic