Ultimaker/Cura #17232: Slicing error

JohnnyDiegem, 2023-11-05:
### Cura Version

5.5.0

### Operating System

Windows 11

### Printer

ultimaker original

### Describe model location

rotated 90°

### Describe your model

Dragon egg Made with fusion 360

### Add your .zip here ⬇️

[UMO_DragonEgg v8.zip](https://github.com/Ultimaker/Cura/files/13260359/UMO_DragonEgg.v8.zip)

JohnnyDiegem, 2023-11-05:
Gives error when printing directly from Fusion 360. When I save the STL to disk and then load it into cura there is no error.

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic