Ultimaker/Cura #17233: Build Plate Temperature Initial Layer is ignored

aspdigital, 2023-11-05:
### Cura Version

5.4.0

### Operating System

macOS Sonoma 14.1

### Printer

Creality Ender 3 Pro (standard firmware)

### Reproduction steps

1. I started with the default Ender 3 Pro "Super" profile.
2. Build Plate Temperature and Build Plate Temperature Initial Layer both have a default value of 60ºC.
3. I changed Build Plate Temperature to 50º and left the Initial Layer to 60º.

### Actual results

Initial build plate temperature, and the temperature for all subsequent layers, is 50º.

### Expected results

Initial build plate temperature should be 60º and all subsequent layers should be 50º.

I looked in the generated gcode for the M140 commands that set build plate temperature.

The first instance is:

```;TIME_ELAPSED:26.860702
;LAYER:1
M140 S50
```

Layer 1's temperature should be 60, not 50.

The only other instances of the M140 command are at the end, where the heaters are turned off.

### Add your .zip and screenshots here ⬇️

[image]

GregValiant, 2023-11-05:
Thanks for the report.
I have never seen Cura insert a temperature line there.  It might have been added by a post processor.
If the temperatures are not called out in the StartUp Gcode then Cura prepends the temperature before the StartUp to avoid cold extrusions.

Use the "File | Save Project" command and then zip the 3mf file and post the zip folder here.

aspdigital, 2023-11-05:
Project is here. Also attached is the gcode output. 

re: post-processor, unless that's something Cura does, I didn't install anything beyond the application itself.

[Rib-r3v2-new (metal project).3mf.zip](https://github.com/Ultimaker/Cura/files/13260603/Rib-r3v2-new.metal.project.3mf.zip)
[Rib-r3v2-new (metal).gcode.zip](https://github.com/Ultimaker/Cura/files/13260619/Rib-r3v2-new.metal.gcode.zip)

aspdigital, 2023-11-05:
There is start G-Code:

```; Ender 3 Custom Start G-code
G92 E0 ; Reset Extruder
G28 ; Home all axes
M104 S{material_standby_temperature} ; Start heating up the nozzle most of the way
M190 S{material_bed_temperature_layer_0} ; Start heating the bed, wait until target temperature reached
M109 S{material_print_temperature_layer_0} ; Finish heating the nozzle
G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
G1 X0.1 Y20 Z0.3 F5000.0 ; Move to start position
G1 X0.1 Y200.0 Z0.3 F1500.0 E15 ; Draw the first line
G1 X0.4 Y200.0 Z0.3 F5000.0 ; Move to side a little
G1 X0.4 Y20 Z0.3 F1500.0 E30 ; Draw the second line
G92 E0 ; Reset Extruder
G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
G1 X5 Y20 Z0.3 F5000.0 ; Move over to prevent blob squish
```

and I can see it's put into the design's gcode file with the correct first layer temperature:

```
; Ender 3 Custom Start G-code
G92 E0 ; Reset Extruder
G28 ; Home all axes
M104 S175.0 ; Start heating up the nozzle most of the way
M190 S60 ; Start heating the bed, wait until target temperature reached
M109 S220.0 ; Finish heating the nozzle
G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
G1 X0.1 Y20 Z0.3 F5000.0 ; Move to start position
G1 X0.1 Y200.0 Z0.3 F1500.0 E15 ; Draw the first line
G1 X0.4 Y200.0 Z0.3 F5000.0 ; Move to side a little
G1 X0.4 Y20 Z0.3 F1500.0 E30 ; Draw the second line
G92 E0 ; Reset Extruder
G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
G1 X5 Y20 Z0.3 F5000.0 ; Move over to prevent blob squish
```

I'm in the habit (good, bad, indifferent, I dunno) of pre-heating the bed to the initial layer temperature (in this case 60º), and when I started the print that uses the different first layer, I saw that it immediately went to the main (not initial) temperature of 50º. This is why I started looking at gcode. I forgot about the start-up code, and the M190 command it sends. But something's weird, as if the M190 command is ignored.

GregValiant, 2023-11-05:
What I see in the gcode is correct and your submitted gcode is correct.  Gcode is numbered in Base0 and you appear to have skipped over LAYER:0.
In shorthand, the bed temperature are:
M190 S60 ; Start heating the bed, wait until target temperature reached
;LAYER:0
;TYPE:SKIRT
;TYPE:WALL-INNER
;TYPE:WALL-OUTER
;TYPE:SKIN
;LAYER:1
M140 S50 ;drop the bed temperature because the Initial Layer is done printing
The Initial Layer is printed at 60 and starting with LAYER:1 the rest of the file is printed at 50.  What I see here is expected behavior so I'll go ahead and remove the bug label and close this.  Let me know if you feel that is premature.

closed 2023-11-05 by GregValiant