Ultimaker/Cura #17234: Slicing failed on ender 3 v2

Sam-Maya, 2023-11-05:
### Cura Version

5.5.0

### Operating System

Windows 11

### Printer

Ender 3 v2

### Name abnormal settings

None - this is a fresh cura install

### Describe model location

does not slice even when moved around the print bed

### Describe your model

Its an electric violin model i got off thingyverse

### Add your .zip here ⬇️

[CE3E3V2_EZ2PRINT01.zip](https://github.com/Ultimaker/Cura/files/13260667/CE3E3V2_EZ2PRINT01.zip)

Sam-Maya, 2023-11-07:
i believe the error is due to the height of the object being too tall.

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic