Ultimaker/Cura #17235: Cura 5.5 crashes adding new printer

jesternberg, 2023-11-05:
### Cura Version

5.5.0

### Operating System

Windows 11

### Printer

none

### Reproduction steps

1. Select Printers drop down
2. Select Add Printer
3. Select Non UltaMaker printer
4. Select Add a non-networked printer
5. Scroll down by clicking scroll bar a few times

### Actual results

Appication crashes, no processes left in Task Manager, no log file in folder

### Expected results

ability to add new printer without crashing

### Add your .zip and screenshots here ⬇️

no relative screen shots.

janbunnik, 2023-11-07:
I have the same problem ion the same sequece

jesternberg, 2023-11-07:
more info. scroll by chick and hold i.e. drag, works. multiple clicks to page down is how to recreate the crash. the crash is 100% repeatable and also occurs in 5.4