Ultimaker/Cura #17238: Slicing failed...

funch1, 2023-11-06:
### Cura Version

5.5.0

### Operating System

Windows 11

### Printer

Ender 5 S1

### Describe model location

Flat, no does not slice even if rotated.

### Describe your model

It sliced fine before adding more height to the top layer in TinkerCad.

### Add your .zip here ⬇️

[FailedSlice.zip](https://github.com/Ultimaker/Cura/files/13262067/FailedSlice.zip)

I included the stl from TinkerCad and the saved 3mf from cura.

thank you!

yzheka, 2023-11-06:
I also have same error. No error message
[image]

911Henrik, 2023-11-08:
Slicing failed in 5.5.0 beta 1 
[image]

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic

funch1, 2024-04-03:
Yes that sliced correctly.   Thank you!