Ultimaker/Cura #17247: Support overhang angle seems to cause error.

mewbusi, 2023-11-07:
### Cura Version

5.1.0

### Operating System

Windows 10

### Printer

Creality Ender 3

### Name abnormal settings

Just reduced the support overhang angle to 50 deg.

### Describe your model

Yes I have sliced the model before and had a successful print.

### Add your .zip here ⬇️

[CE3_Ralsei_CH1.zip](https://github.com/Ultimaker/Cura/files/13273705/CE3_Ralsei_CH1.zip)

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic

mewbusi, 2024-04-08:
Certainly I will go ahead and try that out sometime soon.

mewbusi, 2024-04-08:
It seems to not have the issue anymore but I would really need to probe it
to be sure. I will let you know if it happens again.