Ultimaker/Cura #17253: Slicing error with trees

burkes13, 2023-11-07:
### Cura Version

5.5.0

### Operating System

Windows 11

### Printer

Custom Printer

### Name abnormal settings

No unusual or advanced settings. Does slice when normal support selected, but when trees selected it fails

### Describe your model

Model is of horns, the model can be slices when normal support is selected, not with trees.

### Add your .zip here ⬇️

[CFFFP_JesterHorns1.zip](https://github.com/Ultimaker/Cura/files/13284812/CFFFP_JesterHorns1.zip)

doubletrouble023, 2023-11-08:
I also get this a lot of the time.
Sometimes it works as usual if I uncheck "Create support" and/or "Tree" a few times.

5axes, 2023-12-07:
No issue on my Windows 10 laptop ...  
[image]

The Windows 11 issue ? https://github.com/Ultimaker/Cura/discussions/16440#discussioncomment-7790452

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic