Ultimaker/Cura #17254: Z -Seam on spiralize outer contour

Evgaleks, 2023-11-07:
### Cura Version

5.5.0

### Operating System

Windows 10

### Printer

Custom FDM printer

### Reproduction steps

1. Select  _Print Settings_ - **Special Modes**
2. Select  _Surface Mode_ - **Surface**
3. checkbox - **Spiralize Outer Contour**
4. Start the **Slicing** process

### Actual results

Appears **Z-Seam**
[image]

and there are additional lines in the g-code:

[image]

the seam is also obtained on the finished printed model.

### Expected results

when choosing _Surface Mode_  -  **Normal** (or)  **Both**   
and  _checkbox_ - **Spiralize Outer Contour**
everything works correctly and the **Z-seam** does not appear.
[image]
[image]

### Add your .zip and screenshots here ⬇️

[image]

GregValiant, 2023-11-10:
Thanks for the report.
To me it looks like it's saying you can't do both.  I would agree it doesn't actually say that anywhere but it is the users responsibility to check the preview.
When you switch to Surface Mode maybe it should turn Spiralize off and if you select Spiralize then the Surface Mode should be set to Normal.  It might be doing that internally but that isn't how the setting boxes react.

The Cura team will take a look.

Evgaleks, 2023-11-10:
Hi,

Sorry, but you didn't look at the error message carefully. When  select the  Spiralize checkbox, the g-code file is generated correctly (for all modes “Normal”, “Surface” and “Both”) 

with movement along the spiral path, but only for the Surface Mode  there are erroneous "G0..." lines between the "layers" in the g-code file.

 And this leads to a printing defect.

 

Best regards.

Evgeny

 

This email has been checked for viruses by Avast antivirus software.
www.avast.com

GregValiant, 2023-11-10:
Spiralize generates a continuous extrusion up the model.  Surface mode treats a solid as a surface but with conventional layering.  That is why you are getting the G0 travel moves.  If it is in surface mode then that appears to be overriding the spiralize setting.
So the bug I see here is that Cura should allow either Spiralize, or Surface, but not both together.

Evgaleks, 2023-11-11:
Hi,

 

Thanks for the answer. I apologize for my English, maybe because of this there is some misunderstanding. I just want to clarify some details.

 In general, when the "spiralize" checkbox is enabled, the g-code file is created correctly with continuous stretching of the model upward in all modes - "Normal", "Surface" and "Both",

 except that in the "Surface" mode in the g-code file there are erroneous lines "G0" - (without moving along the Z axis).

 This leads to a defect when printing the model, and the appearance of a Z-seam in the editor.

 

Part of the g-code for printing a simple cylinder with the "Spiralize" checkbox enabled:

 

 

Best regards.

Evgeny

GregValiant, 2023-11-12:
It appears that if you use "Spiralize Outer Contour" and have "Surface Mode" set to "Surface" that there will be a Z-seam.  That is because internally "Surface" is overriding "Spiralize".
If you do not want the Z-seam then do not select them both, just use "Spiralize" in "Normal" mode.
There is always a Z-seam if Surface is selected.  That is not the bug.
In my opinion, the bug is that Cura does not inform the user that "Surface" is not compatible with "Spiralize".  That allows a user to select both of them and as you found, that doesn't work.

Evgaleks, 2023-11-12:
No, this is just a bug because in the “Surface” mode with the “Spiralize” checkbox turned on, the g-code file is created with a smooth stretching of the object along the Z axis. But at the same time, there are erroneous lines with G0 code for moving along X and Y, which create a defect when printing. If you don't think this is an error, try to explain the purpose of these lines with G0 code with absolutely useless movement along the X and Y axes while movement along the Z axis is in a spiral.

Evgaleks, 2023-11-12:
If we talk about the logic of the program, then it is much more logical for the “Spiralize” function to work only in the “Surface” mode. Because in reality, in the “Normal” mode, the function of smoothly pulling the model up physically cannot work without converting the model into a Surface object. So why such confusion? While in the “Surface” mode this is more logical and understandable, it is possible to print the model’s outline both in the normal mode in layers (for complex unconnected contours) and in the mode of smoothly stretching the model upward (for closed contours)

xamiell, 2024-01-15:
Hi!
Any update?

uzmn, 2024-12-20:
I wrote a Python code to fix it instead of waiting for Cura:

import os
filepath = r'C:\Users\UZMN\Downloads\3D printer\gcodes\WT_Roller.gcode' # example gcode

outfile  = os.path.join(os.path.dirname(filepath) , "new_surface.gcode")

fout = open(outfile,'w')

is_spiral_start = False
with open(filepath) as fin:
   for line in fin:
      if is_spiral_start:
         if 'G0 F600 ' not in line and 'G0 F5100 ' not in line:
            fout.write(line)
      else:
         fout.write(line)
      if ';TYPE:WALL-OUTER' in line:
         is_spiral_start = True
fout.close()

GregValiant, 2026-10-25:
I'll go ahead and close this.

closed 2026-10-25 by GregValiant