Ultimaker/Cura #17256: Material Manager - Please add ability to create profiles tiered to specific installed 3D printers

LilBub, 2023-11-08:
### Is your feature request related to a problem?

Right now, when I go into the Material Manager, I have to create long Display Names to indicate which machine the material is set for.
This can be a hassle when you have a print farm with 20 different machines.  
Cura is setup to show Display Name, Brand, Material Type and Color.  
There isn't an option to detail what Machine this material is setup for.

As you can see in my screen shot, I have various Zyltech materials for different filaments and the settings for different machines (Bondtech Eryone Revo, CR-10S, CR-10S Pro, etc).  Each machine is setup differently, for example, direct drive, upgraded hot ends and larger nozzles)

[image]

### Describe the solution you'd like

Perhaps add an entry for MACHINE and this could be a pulldown box which allows the user to select from the installed printers?

This could be the same list of machines found on the menu bar?
[image]

### Describe alternatives you've considered

I've had to use what is there, since Cura's Material Manager seems to be built for only one 3D printer.
With the low cost of machines, most users have more than one 3D printer.

As described earlier, my alternative has been to maximize the Display Name to include [MachineName,Brand,Color] in order to create and maintain material profiles for many machines.

### Affected users and/or printers

Those with multiple machines would benefit for a more robust Material Management system.

### Additional information & file uploads

An example of how my Display Name works in the Manufacturer, Machine and Material type

[image]

5axes, 2023-11-10:
I don't really understand the request ... In Cura it is possible to define a material file and for each material you can define the parameters per machine: Example in my case   FlSun V400  & Creality Ender 3

[image]

In cura For the same material if I Load the FlSun V400  : PLA 208°C

[image]

In cura For the same material if I Load the Creality Ender 3 :  PLA 203°C

[image]

No need to define One material by Machine ???

LilBub, 2023-11-10:
I appreciate the reply but but I don't think you looked at my request in depth

If you do have a work around, you should document this within Cura so that users can examine it.  I haven't seen much in terms of documentation with Material Manager or Print Profiles.

Your remark about "creating my own files" comes across as snarky, indicating it is bothersome for the slicer to have this flexibility.  However other slicers do have this capability, like S3D and others.  I prefer Cura.  :) 

Each material is temperature tested, flow tests and retraction tests for optimal print performance.  Those variable are different on an Ultimaker 2+, a Creality CR-10 S5 with DDX/Slice high flow nozzles, or the Ultimaker S5's.  So you see, having one material setting for "all" really doesn't work.

5axes, 2023-11-10:
It was just to help you. But as a user of 20 different printer you certainly have much more knowledges , sorry for this answer, good kick for your request

rdbischoff, 2024-02-06:
> I don't really understand the request ... In Cura it's possible to define a material file and for each material you can define the parameters per machine: Example in my case FlSun V400 & Creality Ender 3
> 
> ![image](https://private-user-images.githubusercontent.com/11015345/282112105-b5666564-0a75-456b-89b3-46d0f45830cc.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDcwNzc3NDEsIm5iZiI6MTcwNzA3NzQ0MSwicGF0aCI6Ii8xMTAxNTM0NS8yODIxMTIxMDUtYjU2NjY1NjQtMGE3NS00NTZiLTg5YjMtNDZkMGY0NTgzMGNjLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAyMDQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMjA0VDIwMTA0MVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPThjNDZkZGQwNDQ3NDA0NjVlM2JmNjVlMDAwYzA3NzhiMDQwZjVmOTk1MGE2NWYzYzBmMTg0MmU5MmY4YzUxOTgmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.CuKoqV1Yp8Y4bMMIqvv0Ej3WpjsGb36lnKJrJRGTDN4)
> 
> In cura For the same material if I Load the FlSun V400 : PLA 208°C
> 
> ![image](https://private-user-images.githubusercontent.com/11015345/282113056-b27139ad-3ed1-488f-8bb7-25e3602a5b80.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDcwNzc3NDEsIm5iZiI6MTcwNzA3NzQ0MSwicGF0aCI6Ii8xMTAxNTM0NS8yODIxMTMwNTYtYjI3MTM5YWQtM2VkMS00ODhmLThiYjctMjVlMzYwMmE1YjgwLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAyMDQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMjA0VDIwMTA0MVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWM4YTVmNzY1ODg3NTI2ODU3MzhhNWU3ZTcwOGEyNzNkMDRkNTExNmQ3YWM1NmRmODQ4ZjRjNjg1NGUwMjA5ZWEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.85QNl0L-0Mq2Kim3g-x_ffiIpi3fSUHG-YRqje5unHA)
> 
> In cura For the same material if I Load the Creality Ender 3 : PLA 203°C
> 
> ![image](https://private-user-images.githubusercontent.com/11015345/282113364-882b252a-04a6-400f-a7e6-c3a721243524.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDcwNzc3NDEsIm5iZiI6MTcwNzA3NzQ0MSwicGF0aCI6Ii8xMTAxNTM0NS8yODIxMTMzNjQtODgyYjI1MmEtMDRhNi00MDBmLWE3ZTYtYzNhNzIxMjQzNTI0LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAyMDQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMjA0VDIwMTA0MVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTAyYWNjNmExNjFmZWM1MmZkNTI1YmNjNDRmNWM5MTYxNWQxZDM2Y2Q5NzE5NjQ3MjhjZmE5YTRkYTAwMzZmOTgmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.h5QGl3ObAWYJT2XeMbRSvixWKf2Ze53FbSuPcikmHRw)
> 
> No need to define One material by Machine ??? Basic operation already includes the option of defining machine-specific parameters.
> 
> What is true is that if you want to make changes for all the machines using cura, you have to load the machine, change the value in the material (the change will be updated in the machine entry, if it exists in the material file) and do it again for each machine. That's why I prefer to edit the text file directly in my **materials** folder.
> 
> If there's anything we can do to make it simpler, it might be in this area. But, if I had 20 machines to manage, I'd spend some time creating my own files, or even make my own application to manage my settings according to my needs.

Where is the materials folder to find the text file?

5axes, 2024-02-06:
in the Menu : Help -> Show Configuration Folder     then  materials folder to store your own material configuration file (extension .fdm_material)