Ultimaker/Cura #17268: Slicing failed on topo map

Nickdaco, 2023-11-09:
### Cura Version

5.5.0

### Operating System

MacOS Sonoma

### Printer

Ultimaker s3

### Describe your model

Topological map of Washington

### Add your .zip here ⬇️

[washington.stl.zip](https://github.com/Ultimaker/Cura/files/13304051/washington.stl.zip)

GregValiant, 2023-11-10:
Thanks for the report.
In the Cura Marketplace is the "Mesh Tools" plugin.  When it is loaded you will see that your model looks like this:
[image]

The polka-dot surfaces are indicating errors in the model.  I was able to repair the model with MS 3D Builder but it still wouldn't slice.  I finally went back to 5.2.2 and it sliced.

I'll mark this as a duplicate and the Cura team will take a look.

Nickdaco, 2023-11-14:
Thanks for checking it out!

Do you know why it would be slicing completely fine on my Ender-3 presets? When I slice it with the same settings but my printer is set as an Ender-3 I have no problem.

Also PrusaSlicer has no problem slicing it, so I thought there wasn't a problem with the STL.

GregValiant, 2023-11-14:
There are some models that show up here that Cura 5.3 and above simply will not slice (Moon Lamps and Crystal Dragons come immediately to mind).
Then there are models that won't slice if they are at the midpoint of the Cura build plate but moving and/or rotating and/or changing the Initial Layer Height by just a little bit (like 0.03) can get them to slice.
My guess (and it's only a guess) is that the way the slicing engine "sees" the model has something to do with this.  Changing the Initial Layer Height moves the cutting plane in the Z so it goes through the model at a slightly different place and maybe avoids whatever was giving it heartburn.  Moving in the X or Y and rotating the model have the same sort of effect within the cutting plane.
There is another sub-set of models (generally buildings) that have things like rough tile floors or rough-hewn planks on the floor.  Those have about a bazillion triangles and vertices extremely close together in the Z so they end up with about half a bazillion "islands" that need to be calculated during the slice.  Maybe that just overwhelms the engine.  I dunno.

It's troubling and I'm sure it's being worked on but so far it remains an issue through 5.5.0.
In an oddity, many of the models that others can't slice - I don't have any trouble with.  It's annoying for both the poster (I've been accused of having a "special" version of Cura) and for the Cura team because if you can't get a good handle on a problem, it's really hard to fix.  A consistent problem that anyone can duplicate is way easier to troubleshoot.

I trust 4.13.1 the most and 5.4.0 the least.  5.5.0 shows a bit of improvement over 5.4.0 but it certainly isn't right yet.  Fortunately we can have more than one version installed at the same time.  People have asked for a switch so they can turn the Arachne engine on and off alla PrusaSlicer.  My "switch" is flipping from 4.13.1 to 5.2.2.  

On the plus side, Cura remains highly customizable.  The plugin system gives people a chance to alter a gcode and tune it in very nicely.  Of course if the model won't slice, who cares!

GregValiant, 2026-10-01:
I'll close this as stale.

closed 2026-10-01 by GregValiant