Ultimaker/Cura #17270: WCF_Top

EstebanClaro, 2023-11-09:
### Cura Version

5.5.0

### Operating System

Windows 11

### Printer

artillery genius

### Describe your model

Trophy World Cup Female (top part)

### Add your .zip here ⬇️

[AG_WCF_TOP.zip](https://github.com/Ultimaker/Cura/files/13307957/AG_WCF_TOP.zip)

EstebanClaro, 2023-11-09:
Bottom part of this model (brokendown into two pieces by means of MeshMixer) was sliced properly

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic