Ultimaker/Cura #17271: Insane amount of material added to sliced model

ValerioMalara, 2023-11-09:
### Cura Version

5.4.0 - 5.5.0

### Operating System

Windows 10

### Printer

FLSUN SR

### Reproduction steps

Tried to print a model with hollow parts.
[image]

### Actual results

Material was added to the sliced model, certain orientation seens to aggravate the issue
[image]

### Expected results

Correct slice of the model with no added geometry

### Add your .zip and screenshots here ⬇️

[FSR_Frame-Katana.zip](https://github.com/Ultimaker/Cura/files/13308142/FSR_Frame-Katana.zip)

chipmunk1977, 2023-11-09:
This looks like you might have got the parameter "make overhangs printable" selected. If so, I'd try slicing it with that unchecked.

GregValiant, 2023-11-10:
That's what it looks like to me as well.
@ValerioMalara was that the problem and can we close this?

closed 2023-11-25 by github-actions[bot]