Ultimaker/Cura #17275: T0 inserted before startup script

timrhallam, 2023-11-09:
### Is your feature request related to a problem?

The initial "T0" that Cura inserts before the startup script is causing me issues, and I seem to be in good company according to a day's worth of searching for a solution. On my machine (based on the E3D toolchanger running Duet 3), homing MUST be carried out before selecting a tool, otherwise bad crunchy things can happen such as crashing into the bed on z-probing or trying to pick up a tool without knowing where the head is.

### Describe the solution you'd like

If "T{initial_extruder_nr} is not found in the startup script, insert T0 as currently happens, otherwise don't add any T commands before startup.
A similar logic is already implemented for heaters, which works perfectly.

### Describe alternatives you've considered

Currently I have to use conditionals to ensure homing has occurred before running any tool-changing macros.
What this means in practice is that:-
If already homed, machine collects T0 (before startup script) only to immediately return it (during startup script), home, then collect again - this is very annoying and slow, but safe.
If not already homed it pretty much ignores the initial T0 because it's closely followed by a T-1 in startup and movement hasn't been allowed to occur in the intervening period.
It's an unnecessary and complex workaround for a line that could just be optional (pretty please).
To the best of my knowledge, this workaround is only available on RepRap firmware due to the implementation of conditional gcode in RRF3.x.
Other people with this issue seem to be manually removing the "T0" from the file after slicing, which is disruptive to workflow.

### Affected users and/or printers

Anyone running multiple tools that might want more specific control of the order of startup operations. Printers such as E3D toolchanger where the Z-probe is higher than the nozzle.

GregValiant, 2023-11-10:
This is very "machine specific".  Most printers it is the other way around.  They want to know which tool to start with.
Since I don't see UltiMaker giving this a high priority I will pass along my re-write of Search and Replace.  There is a pull request in for this.
The current Search and Replace will go through a file and replace _every instance_ of the Search criteria.
Amongst other changes - the re-write will allow you to only replace the first instance it comes across as well as include the StartUp Gcode and Ending Gcode in the searches.

Unzip the file and put SearchAndReplaceII.py into the "scripts" folder within your Configuration Folder.  It will be available in Cura under "Extensions / Post Processing / Modify Gcode" and then "Add a script" and "Search and Replace II".  The original "Search and Replace" will be there as well.
[SearchAndReplaceII.zip](https://github.com/Ultimaker/Cura/files/13320461/SearchAndReplaceII.zip)

This works and replaces just the first instance of "T0" with "**;T0**" making it a comment instead of a command.  You could also replace it with an empty string.
[image]

timrhallam, 2023-11-10:
That's a great improvement @GregValiant. I'll certainly be using it. Thanks so much.
I agree that this is a pretty niche problem at the moment, but one that I feel will become more widespread as toolchangers become more popular as they get developed my mainstream manufacturers. That said, I do understand the tough calls that need to be made by Ultimaker when deciding how much resource to throw at an issue that may or may not become a future problem.
Sadly I don't have the familiarity with how Cura works 'under the hood' to write a solution myself, and I did spend a decent length of time trying.
Following the logic of how the temperature commands are found was simple enough, then I hit a wall where the magic engine fairies appear to take over. I can't work out where "material_bed_temp_prepend" and "material_print_temp_prepend" settings are interpreted after they get dropped into the stack by StartSliceJob.py - to the uninitiated like me, this becomes a needle in a haystack since there's no function-return path to follow.
Any advice would be welcomed - I'd be delighted to contribute to improving a product I use every day for the benefit of everyone that might experience the same or similar issues in the future.

GregValiant, 2023-11-10:
I'm an amateur hack and I'm OK with the post-processors (Remco Burema just rolled his eyes and gagged) because I can start with a clean piece of paper.  Cura is a mix of C++, Python and QML and I don't get it either.  Heh, you figured out that it was the "prepend" that is the problem.  I'll give you points for that.

Scissors95, 2024-04-06:
Your best bet is to switch to another slicer. The devs have no intention of implementing a fix for this since it doesn't affect Ultimaker machines directly.

It sucks because cura is a really good slicer other than making some machines stab themselves in the face.

GregValiant, 2024-04-06:
A couple of years ago a member of the UM Team made a response to a comment like that.  What he said was...
_"There is only so much we are willing to give away for free."_
That makes good sense to me.  They are in the business of manufacturing and selling printers.  Why should they pay their people to support my Creality printer?

Cura is open source and anyone can write a script and customize Cura.  In this case it was a re-write so added options allow the user to replace only the first instance of the search criteria, to include the StartUp and Ending Gcode, to limit the replacements to a range of layers.
I'll post it again.
[SearchAndReplace_GV.zip](https://github.com/Ultimaker/Cura/files/14895212/SearchAndReplace_GV.zip)
Unzip it and put the "SearchAndReplace_GV.py" file into the "scripts" folder in your configuration folder.

There are two ways to skin this cat.  Both will work regardless of which extruder the print starts with.
You can have the "T" line commented out:
Search = T(\d*)
Replace = ;T\1
Regex = True
Replace first instance only = True
Ignore StartUp Gcode = False
or if you just want to add G28 before the tool is picked:
Search = T(\d*)
Replace = G28\nT\1
Regex = True
Replace first instance only = True
Ignore StartUp Gcode = False

Or you can use another slicer.  It don't matter.