Ultimaker/Cura #17277: Failed large print slicing

halfeyedjack, 2023-11-10:
### Cura Version

5.5.0

### Operating System

Windows 10

### Printer

Neptune 4 max

### Describe your model

power ranger helmet

### Add your .zip here ⬇️

[helmet_front-support2.zip](https://github.com/Ultimaker/Cura/files/13315027/helmet_front-support2.zip)

halfeyedjack, 2023-11-10:
fails to slice 9/10 times. when it is successful, print fails with offset layers (each layer increased y direction by ~1mm)

GregValiant, 2023-11-10:
Thanks for the report.
I don't have the definition file for the Neptune 4 so I used the 3 Max and didn't have any trouble slicing it.  There is a bug that can keep models from slicing.  Have you tried moving or rotating the model in Cura?  Sometimes changing the Initial Layer Height by a little bit can fool the bug.
I'll mark this as a duplicate and go ahead and close it.  I hope you understand.

The other issue of layer shifting might be caused by Accel settings that are too high for your printer.  Loose belts can cause the problem as well.  I've investigated hundreds of Cura gcodes and the problem will not be the gcode toolpath.
At 113 hrs this is a long print.

closed 2023-11-10 by GregValiant

halfeyedjack, 2023-11-10:
Thank you for your prompt reply! I have rotated the model and changed layer lines. I completed a similarly large print last night with no issues - the problem seems to be restricted to this file specially within cura

GregValiant, 2023-11-11:
This bug has been present since 5.3.1.  I'm guessing it's an "unintended consequence" of a fix or an addition to Cura.  It is most prevalent on Windows machines but that might be because there are more Windows machines.

An oddity is that I rarely have the problem even with the models that show up here in bug reports. 
So it might be the file...but it might not.  I'm positive about that.