Ultimaker/Cura #17284: Spiral mode adds 6 extra inner walls every 2 layers?

FractalEngineer, 2023-11-10:
### Cura Version

5.5.0, 5.4.0

### Operating System

Win 10 & 11, on quadro P2000 and geforce 4070

### Printer

Railcore II ZL

### Reproduction steps

1. Open any part
2. Pick any profile
3. Select "spiralize over contour"

### Actual results

Every second layer shows 6 unwanted inner walls on top of the set wall line count

### Expected results

There should be only the specified wall line count

### Add your .zip and screenshots here ⬇️

[image]

GregValiant, 2023-11-10:
Thanks for the report.
That is the "Expected behavior" of the 5.x versions of Cura.  You can resubmit this as a feature request if you like.
I'll remove the bug label and close this as what you see is how it is supposed to work and not a bug.  I hope you understand.

closed 2023-11-10 by GregValiant

FractalEngineer, 2023-11-10:
by expected behavior do you mean that it was meant to do that on purpose? 

How is it more desirable than the alternative?

GregValiant, 2023-11-11:
Yes, it was designed in.  Structurally it traps the skins.  That must have been important to someone  I have no idea why it was changed but Spiral in the 5.x versions is different than in the 4.x versions.  More than just the bottom skins was changed.