Ultimaker/Cura #17286: Merged part when multiplying

Brumcm, 2023-11-10:
### Cura Version

5.5.0

### Operating System

macOS Ventura 13.6

### Printer

Ceality Ender 3 Pro

### Reproduction steps

- Download file from https://www.printables.com/model/132643-aqara-temperature-sensor-wall-mount
- Import .STL file
- Select item, multiple for 9 copies
- Select Top View
- Note that piece in the center is a mutant combination of two units

[image]

### Actual results

mutant double piece created

[image]

### Expected results

Should have printed 10 units, not 8 and 1 mutant double piece

[CE3PRO_Aqara_Sensor_Mount.3mf.zip](https://github.com/Ultimaker/Cura/files/13323766/CE3PRO_Aqara_Sensor_Mount.3mf.zip)

### Add your .zip and screenshots here ⬇️

Cura log file...

[cura.log.zip](https://github.com/Ultimaker/Cura/files/13323799/cura.log.zip)

GregValiant, 2023-11-11:
Thanks for the report.
Under "Preferences / Configure Cura" on the General tab you will find "Ensure models are kept apart".  Make sure that is ticked and try it again.
[image]

[image]

I'll remove the bug label.  Let us know if that works for you and if it doesn't I'll put the bug label back on.

Brumcm, 2023-11-11:
Greg,

Thanks for the quick reply!  That did resolve the issue, but I had to clear the build plate, re-add the object and then re-multiply.  Shouldn't that option be checked by default?

thanks!
Bruce

GregValiant, 2023-11-13:
If it is enabled by default then the user can never overlap models as they always jump away from each other.  I have a feeling that a lot of users would complain about that (including me).

If you want to add an override to your definition file you can try.  The setting is named "physics/automatic_push_free".
It is defined in PlatformPhysics.py on line 40.

I'll go ahead and close this as it is "expected behavior" given the setting that you had.  I hope you understand.

closed 2023-11-13 by GregValiant

Brumcm, 2023-11-15:
Understood, thanks for letting me know