Ultimaker/Cura #17287: Native CAD Plugin

darkside8613, 2023-11-11:
### Cura Version

5.5.0

### Operating System

MacOS 14.1.1

### Printer

Ultimaker S5

### Reproduction steps

Open Cura

### Actual results

Error "The plugin NativeCADimport could not be loaded. Re-installing the plugin might solve the issue."

### Expected results

Plugin should load correctly

### Add your .zip and screenshots here ⬇️

none

GregValiant, 2023-11-11:
Thanks for the report.
If the author of the plugin has not updated it to work with the new dialog boxes then it is simply out-of-date.  That's what happened with the SolidWorks plugin.
I'm unfamiliar with the NativeCADimport plugin.  @MariMakes can you offer any insight here?

GregValiant, 2026-10-01:
I'll close this as stale.

closed 2026-10-01 by GregValiant