Ultimaker/Cura #17288: NativeCAD Plugin incompatible with Solidworks 2023

RSchnabel78, 2023-11-11:
### Cura Version

5.5.0

### Operating System

Windows 11

### Printer

Ultimaker S5 latest Firmware

### Reproduction steps

Open native Solidworks 2023 CAD files creates failure message. Also reported in #15163

### Actual results

Not able to load file ; workaround by saving files to STL required.

### Expected results

Using sucessfully SOLIDWORKS2023 files without converting it

### Add your .zip and screenshots here ⬇️

[image]

fieldOfView, 2023-11-11:
Since this is indeed a duplicate of https://github.com/Ultimaker/Cura/issues/15163, I am closing this issue in favor of that previous report.

closed 2023-11-11 by fieldOfView

RSchnabel78, 2023-11-11:
But is there a short term solution available?