Ultimaker/Cura #17290: Unexpected slicing error

codehero, 2023-11-11:
### Cura Version

5.5.0

### Operating System

Linux MINT

### Printer

Anycubic

### Describe your model

Used openscad, is watertight.

### Add your .zip here ⬇️

[CURA.zip](https://github.com/Ultimaker/Cura/files/13325887/CURA.zip)

GregValiant, 2023-11-13:
Thanks for the report.
I didn't have any trouble slicing this on my Windows system.
Try moving it around or rotating it.  Sometimes changing the Initial Layer Height by a little bit can fool this bug.

I'll mark this as a duplicate report and go ahead and close this one.  I hope you understand.

[image]

closed 2023-11-13 by GregValiant

GregValiant, 2023-11-14:
Your model is pretty simple and rectangular.  Most of the models that are affected by this bug are more organic shapes.
Whatever is going on it's been around since 5.3.1 and although the team seems to have managed some improvement, the bug certainly has not been smooshed yet.  In regards to this slicing bug, 5.5.0 is barely better than 5.4 which quite frankly was pretty bad.  But other than this "I ain't gonna slice this" bug 5.5.0 is stable and useable.  But the whole point is to slice things and so for a lot of people it misses the point.
I just revisited your model and yep, it sliced in a hurry and with no problem in 5.5 on my Win 10 system.

codehero, 2023-11-14:
I appreciate a response, but you being to slice my model does not help me. This isn't about particular models, it's about fixing the installation on my system. I need to know more debug info to fix my problem. That includes what command line arguments are passed to the backend slicer as well as debug output from it and possibly Cura frontend itself. I am willing to dig deep but my only troubleshooting recourse seems to be to submit an AMF, which discards all context related to the bug.

GregValiant, 2023-11-14:
@fieldOfView and @Asterchades are more suited to help with that sort of debugging.  They might be able to point you in the right direction.  A person familiar with Linux and Cura Engine would seem to be necessary.  I'm not either of those.

Asterchades, 2023-11-15:
Much as I appreciate the vote of confidence, so far as I know there shouldn't be any appreciable difference - at all - between the slicing engine functionality regardless of the OS and processor architecture. There shouldn't even be any driver interaction at that point as it's all CPU math. All I can really think of is perhaps a compiler issue but surely that would have impacted earlier versions the same.

This will require someone more intimately familiar with Cura than me to diagnose.

codehero, 2023-11-15:
Thanks for your responses so far. Let me ask another way: how would a developer easily monitor console output and shell invocations? Like is there a debug mode in the GUI that would show debug traces from it and the slicer process?