Ultimaker/Cura #17294: Slicing Hangs around 30%

gh13, 2023-11-12:
### Cura Version

5.5

### Operating System

Windows 10

### Printer

Ender 3 S1 PRO

### Name abnormal settings

Seems to be related to the countersunk holes in the base as if I remove the countersink leaving the holes the model will slice

### Describe model location

Centred on build plate

### Describe your model

Model is watertight, no errors, Can slice same model without the countersunk holes

### Add your .zip here ⬇️

[Boost Fork Mount (Van) - [Angles] (Squared) V2 [Flat Mount] [Screw Holes] V5.zip](https://github.com/Ultimaker/Cura/files/13328350/Boost.Fork.Mount.Van.-.Angles.Squared.V2.Flat.Mount.Screw.Holes.V5.zip)

GregValiant, 2023-11-13:
Thanks for the report and the file.
I didn't have any trouble slicing this.  There is a bug in Cura that can sometimes be fooled by moving, rotating the model and/or changing the Initial Layer Height by just a bit.
I'll mark this as a duplicate report and go ahead and close this one as the behavior has been noted.  I hope you understand.

[image]

closed 2023-11-13 by GregValiant

gh13, 2023-11-13:
Hi Greg. All very strange. I just tried opening the file & slicing straight away & it hung slicing.
The bizarre thing is that I’ve just checked a few different support scenarios as I was thinking that the blind holes may be the cause & it will slice if I use Tree supports but not using Normal supports.

Regards Graeme

[Image removed by sender. image]<https://user-images.githubusercontent.com/64202104/282518159-c2962431-e9dc-4525-b412-633ad1250d63.png>

GregValiant, 2023-11-13:
I took another look at this.  I print mostly engineering stuff myself.
I set out trying to get it to not slice for me and I haven't been able to.  I'll set that aside for a minute.
Your print with the tri-hex infill at 75% is a 17hr15min print.

This is grid with the line directions at [0] (infill at 90° to the walls), 12% density and Infill Line Multiplier at 3 (like internal ribs), Infill Layer thickness at 2X layer height (infill goes down every other layer).
It's a 9 1/2hr print and you can park your car on it.
[image]

This is your infill configuration
[image]

I'd put support blockers at the tops of the blind holes.  It will bridge well enough and no one will ever know anyway.

gh13, 2023-11-13:
Hi Greg. Thanks for looking at this again.

This is my first time designing & printing something that requires quite a high engineering strength. I have an engineering background but I’m pretty new to the world of 3D printing & still trying to get to grips with the multitude of settings and their impacts on the resultant print, especially those that would give the best strength, without using materials like abs/asa & having to deal with the added environmental requirements that go with printing with those materials.

The design is intended to be used to mount the forks of a mountain bike with the front wheel removed to my existing truck bed bike rack & because of the orientation I felt that I would have print the design it has the potential to experience tension loading in the z-axis so any tips on optimising z-axis strength would be welcome. Would adding line multipliers, provide better tension strength along the z-axis than using higher infill density?

My original prototype, printed with a 50% cubic infill without the screw holes did perform the intended function of holding the forks very securely but suffered a fracture between layers when I simulated some of the potential loadings. This occurred in the large fillet beneath the horizontal hole that holds the forks.
As a result; In this version I upped infill density & changed the pattern in the hope that would improve the z-axis strength. The addition of the screw holes is one of few hybrid designs I’ve been playing around with, which uses machine screws to add reinforcement to the layers.
Some of the designs I’m trialling have threaded blind holes like this one & others have clearance holes all the way through the design, which would be secured either end & may be the way to go. I might even have to make a metal baseplate to link these screws to the truck bed mounting points, as that might end up being a weak point once I’ve reinforced the main body of the design

Tomorrow when I’m at the computer with cura installed; I’ll get chance to have a look at some of your pointers below,.

Thanks again for your support. Regards Graeme

[image]<https://user-images.githubusercontent.com/64202104/282570997-e5b518b1-d3b4-48a5-8e8d-9ce21305813f.png>

[image]<https://user-images.githubusercontent.com/64202104/282571347-05842dfb-726e-4e1e-b4a3-30ecde54c052.png>

GregValiant, 2023-11-13:
Layer adhesion will always be an issue with FDM printed parts.  If the bike will wobble back and forth then that bracket will alternate between tension and compression from the left end of the axle to the right.
Your idea of putting bolts through it is a good one.  Maybe some sort of U-bolt arrangement to hold the axle down instead of trapping it with the plastic.  The lower portion of your bracket could stay in compression.

You can use "Pause at Height" to stop the print and drop parts in.  If you were to put nut sized hex cavities near the top of the screw holes coming up the from the bottom the whole thing would be a lot more robust.  The clearance hole has to extend past the nut a little bit so the fitment doesn't have to be so precise.

I printed these shark fins for the top of the van so we could find it in parking lots.  There are 4 nuts imbedded (visible along the bottom) and a 3/16 x 9 reinforcement rod inside (it's on an angle).  There were a lot of pauses in this.

[image]

It's all just problem solving.

gh13, 2023-11-14:
Hi Greg. That shark fin is genius & very patriotic too. Fantastic.

Using the Pause at Height feature to embed some fixings sounds like it’s something that warrants further investigation. I’d heard the term referred to but that was about all I knew about it & it could be a viable alternative to having a fixing hole through the top of the print.

Like with software design I guess there’s so many ways to achieve the same end result. Some over-engineered & bloated, some under developed & full of bugs & then there’s the goldilocks solution, which is mmm just right 😊

I’ve got quite a few fixings available so I’m going to try some configurations on the prototype that had the failure. If I can strengthen the area around the failure allowing it to be used then that will give me a solid direction for the V2

Cheers Graeme

[Image removed by sender. DSCN3174]<https://user-images.githubusercontent.com/64202104/282598208-8ab70170-2e94-46de-b3fa-c06900c8e9b6.JPG>

GregValiant, 2023-11-14:
Rapid prototype.  Putting our time-involved aside, mistakes only cost a couple of cents.
Some things you can anticipate.  I knew that the Florida sun would be tough on roof mounted plastic things.  I used PETG and the steel backbones keep down the warpage.
These are the gray shark colored fins.  This was at the pause for the nuts.
The "elephant ears" are designed in and get nipped off.  PETG wants to warp off the build plate and those features help hold it down.
[image]

Pausing and resuming a print takes a little practice.  I don't use the retract in the Pause At Height script.  I just hand feed the new filament in until I'm happy with the color coming out of the nozzle.  I grab the booger with tweezers and then hit the button on the LCD and off it goes.  If the "resume" is within either support or infill you never see the resume point.  If it is on an outer wall there can be a mark.

gh13, 2023-11-14:
Hi Greg. That’s great seeing the fixings in-situ mid print. I’ve been using the ‘Tab’ &  ‘Tab Plus’ Cura plugin’s to generate mouse/elephant ears for PETG prints when I dodn’t want to print with a brim. Functionally they do the job well although the plugins do have a couple of usability quirks that require some profile settings changes so they generate properly when sliced.
Likewise with your reasons for printing with PETG, which although is not so much a problem in the UK, when I’ve been on my travels in Europe I’ve often been confronted by what can only be described as a blast furnace upon opening the van after a day in the sun

I did take a quick look at the Pause at Height script following your previous message & wasn’t sure how printing would be resumed following the pause. From your description though it seems to be similar to resuming after a filament run-out in that it’s done via the LCD but without the filament purge feed that the printer performs prior to resuming the print.

Cheers Graeme

[DSCN2809]<https://user-images.githubusercontent.com/64202104/282899879-b027e3cb-3685-415c-83a5-235a17368597.JPG>

gh13, 2023-12-23:
Hi Greg. I’ve been road testing my strengthened bike fork holder for the last couple of weeks & quite happy that it’ll do the job nicely. 😊 In the end I didn’t go down the pause at layer route to insert some fixings as I printed it while I was at work so my design took another direction with some through fixings. Here’s a couple of pic’s of the installation.
Thanks for your advice & guidance on the multitude of options open to me.
Have a great Christmas.

Graeme