Ultimaker/Cura #17297: Moonlamp slicing error

iakten, 2023-11-12:
### Cura Version

5.5.0

### Operating System

windows 11

### Printer

ender 3 pro 

### Name abnormal settings

I haven't tried with default settings

### Describe model location

no to both

### Describe your model

no this is the first time, yes it should be air tight. no I haven't tried mesh fix. 

### Add your .zip here ⬇️

It is fairly used model in thingiverse 
https://www.thingiverse.com/thing:2771919

project file
https://we.tl/t-wOTiUoYEfI

GregValiant, 2023-11-13:
Thanks for the report.
This is the 5th or 6th time a moon lamp or moon model has been reported as not slicing.  I'll leave this for the Cura team.
Out of curiosity - what moon is that?  It doesn't look like Luna.

iakten, 2023-11-13:
No idea, i thought it was Luna

İlkan Akten

şunu yazdı:

GregValiant, 2026-10-01:
I'll close this as stale.

closed 2026-10-01 by GregValiant