Ultimaker/Cura #17298: Slice failed

Pawel-sp9pw, 2023-11-12:
### Cura Version

5.5.0

### Operating System

Windows 10

### Printer

Ender 3 s1 plus

### Describe your model

After scale fails to slice

### Add your .zip here ⬇️

[CE3S1P_hogwards_bookmark.zip](https://github.com/Ultimaker/Cura/files/13328888/CE3S1P_hogwards_bookmark.zip)

GregValiant, 2023-11-13:
Thanks for the report.
When I opened the file the model was off the build plate.  When I moved the model onto the build plate it sliced.  I am unable to duplicate this behavior.  
Have you tried moving and/or rotating the model?  Maybe change the Initial Layer Height by a little?

Pawel-sp9pw, 2023-11-13:
I've tried to move and rotate but no results..
[image]

gazhay, 2023-11-15:
Can confirm - models open and slice fine with 5.1.0 but 5.5 "slice failed" and gives an empty list of settings which may have caused the error.

GregValiant, 2023-11-15:
@gazhay  what do you meany by _"...and gives an empty list of settings..."_?

Pawel-sp9pw, 2023-11-16:
I think i known where is the problem. 
It can slice when i change slice profile. 
I use custom where that one project can't be sliced but when i change to standard (build-in profile) it slice as normal. 
Custom profile that i use [](https://thangs.com/designer/CHEP/3d-model/Ver3.2%20Cura%20V5.0%20Filament%20Friday%20ExtraFast%20%280.28%29%20Profile-942351)