Ultimaker/Cura #17308: Creality CR10-SE Support

Selv96, 2023-11-13:
### Is your feature request related to a problem?

I've been printing on a CR10-SE but getting bad results when using Cura. Using the same profile and slicing with Creality Print it yields a much cleaner finish. I think this is due to using the wrong printer settings in Cura (because there isn't a default for the CR10-SE)

### Describe the solution you'd like

Default Printer settings for CR10-SE

### Describe alternatives you've considered

Different Slicing software

### Affected users and/or printers

Creality CR10-SE

GregValiant, 2023-11-13:
The definition files come from the community or from the manufacturer.  When someone submits a Pull Request for the printer it will show up on the Pull Request page here on Github.  It would then await approval from the Cura team.

GregValiant, 2024-01-02:
This line in the StartUp won't work:
M109 S[material_print_temperature_layer_0]
because Cura uses curly brackets to denote replacement patterns.  It should be:
M109 S{material_print_temperature_layer_0}
Other than that it looks like the boilerplate StartUp Gcode for any Creality printer.  Even if there are some differences between the SE and a regular CR10, I'd just use the CR10 definition.  You can change any settings in the Machine Settings window to be whatever the SE requires.

Theseustark, 2024-02-06:
Which ones should I enter in the starting g code and ending g code sections and where should I set them? Can you help me? I don't know much about this subject.

wim2008, 2024-08-14:
The gcode for CR-10 SE is 99% ok. 

See topic from rremo.
This line in the StartUp won't work: M109 S[material_print_temperature_layer_0] 
It should be: M109 S{material_print_temperature_layer_0}

After that everything is oke for this gcode.
Good luck.