Ultimaker/Cura #17311: Slicing Failed for High Detail Model

nanamiku739, 2023-11-14:
### Cura Version

5.50

### Operating System

Windows 10

### Printer

Creality CR-10S5

### Name abnormal settings

Using slightly altered default profile

### Describe model location

Center of the buildplate

### Describe your model

Slicing always failed. Tried to simplify the model with Prusa Slicer but still the same

### Add your .zip here ⬇️

[Skull2.zip](https://github.com/Ultimaker/Cura/files/13345447/Skull2.zip)

GregValiant, 2023-11-14:
Thanks for the report and the file.
I used Cura's Mesh Tools and the model is not watertight.  There are a couple of other errors as well.  This is in Mesh Mixer.  Each red and magenta ball indicates an error in the model.

[image]

This report is from formware.co/onlinestlrepair:
-> Analysed your file:
--> 0 Naked edges (?)
--> 0 Planar holes (?)
--> 0 Non-planar holes (?)
**--> 11567 Non-manifold edges (?)
--> 14651 Inverted faces (?)**
--> 0 Degenerate faces (?)
--> 0 Duplicate faces (?)
--> 0 Disjoint shells (?)
-> Repairing: 100.00%
----- Repair completed in 20579ms ------
**-> Vertex count changed from 71017 to 60448 (-10569)
-> Triangle count changed from 156262 to 125156 (-31106)**

There is no guarantee that a repaired model will slice (5.5.0 wouldn't slice this) but without repair, I wouldn't trust this model to print correctly.

On another note, you are using Tree Supports with a density of 10% and no external walls.  Some of the supports won't form.  You would be better off with 1 external wall and 0% density.  That would insure that any small supports could at least grow.
This is in 5.2.2.  You can see that it's a 132 hour print and 805gr of filament.  4.13.1 makes it a 187 hour print and in PrusaSlicer it's a 149 hours.
[image]

I'll mark this as a duplicate report as it won't slice in 5.5.0.  I'll leave this for the Cura team to look at since it's another organic shape that won't slice.

nanamiku739, 2023-11-15:
Thanks for the help. I tried repairing it with Prusa Slicer to make it manifold. The repair was successful and Prusa Slicer can slice it. But when I tried to slice the repaired file with Cura 5.50, slicing failed. I even tried to simplify the model in Prusa to make it more low poly, but it still failed in Cura

GregValiant, 2023-11-15:
This bug showed up with a vengeance in 5.3.0.  Since it didn't exist in 5.2.2 I would guess that the bug is an unexpected breakage due to something else that was done/added to the code for 5.3.
5.4 was worse and 5.5.0 is a slight improvement but the bug has not been stomped yet.

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic