Ultimaker/Cura #17313: Falha Fatiamento - snowspeeder

cesarabud, 2023-11-14:
### Cura Version

5.5.0

### Operating System

Windows 10

### Printer

Creality Ender-3 Pro

### Name abnormal settings

Yes

### Describe model location

No

### Describe your model

Yes

### Add your .zip here ⬇️

[3-fuselaje.zip](https://github.com/Ultimaker/Cura/files/13348572/3-fuselaje.zip)

cesarabud, 2023-11-14:
Thanks, Greg.
I was able to slice the model with the file you attached

GregValiant, 2023-11-14:
After I posted that I exported that model from Cura as an STL and imported it into MS3D Builder.  Doing that causes the block and model to merge.  After saving in Builder and reloading in Cura the problem was gone.  I'm not letting Cura off the hook on this but it is an oddity that merging .0144mm³ of mesh into that spot made the difference.