Ultimaker/Cura #17317: German translation error in message on operation to put an object flat on the table automatically

gluetolf, 2023-11-14:
### Cura Version

5.5.0

### Operating System

Windows 11

### Printer

Ultimaker 3

### Reproduction steps

1. Open model in Cura
2. Use method from rotate to auto align model flat on table
3. See the translation error in the message that is shown during the operation

### Actual results

Message reads in German:
"Objekt flach auf die Druckplatte le**b**en ..."

### Expected results

Correct message:
"Objekt flach auf die Druckplatte le**g**en ..."

### Add your .zip and screenshots here ⬇️

[image]

MariMakes, 2023-11-16:
Welcome to the UltiMaker Cura Github 🚀 
Thanks for your report 👍 

I can update the text for you, if it passes the review, you can see the fix in the next release.
Fingers crossed 🤞

casperlamboo, 2023-11-27:
Thanks for your contribution, the translation is updated and should be updated in our next release (5.6)

closed 2023-11-27 by casperlamboo