Ultimaker/Cura #17318: give us a 1.0mm nozzle >:(

REDACTED-O5-13, 2023-11-15:
### Is your feature request related to a problem?

Im always frustrated when I go to attempt to print with my elegoo Neptune 3 pro and I can't print with a 1.0mm nozzle, if we are able to print with as many nozzle sizes as we need I believe this will make everybody happy

### Describe the solution you'd like

this will solve not being able to print in 1.0mm nozzle diameter

### Describe alternatives you've considered

no alternatives just give us the nozzle size

### Affected users and/or printers

everybody will benefit from this, all Neptune 3 pro users will be happy

GregValiant, 2023-11-15:
The nozzle sizes are part of the definition files and those come from either the manufacturer, or from the community members.  For example all the Creality printers include the definition files for 1.0 nozzles and I'm sure a couple of other manufacturers do as well.  This is an Elegoo issue rather than an UltiMaker Cura issue
You can create your own nozzle file by opening one of the existing ones for your printer and changing a couple of lines.  Then do a save-as with the same filename format.  The nozzles are in the "variants/elegoo/elegoo3pro" folder in the Cura installation folder.

LilBub, 2023-11-16:
Agree.  Just change the bead size manually.

REDACTED-O5-13, 2023-12-27:
> The nozzle sizes are part of the definition files and those come from either the manufacturer, or from the community members. For example all the Creality printers include the definition files for [...]

you see im an idiot so youre basically gonna have to walk me through this, ill just ask a bunch of questions
do i have to copy a nozzle file then it creates a new one?
what lines would i have to change for the 1.0mm nozzle? what are the best specifications?
does this apply to mac as well?
what file type is it

GregValiant, 2023-12-28:
You can open the nozzle files in any text editor.  The entire file contents is about 10 lines long.  I'm sure you can figure out which two numbers to change.  Then do a save-as using the same file naming format.

The nozzle size dictates a couple of defaults (like the Line Width, Minimum Line Width, and Layer Height).  You can enter those manually into Cura.  Cura has no idea what nozzle size is actually in the printer so it doesn't care if you have a 0.2 nozzle listed in Cura with a .5 layer height and 1.0 line width.

This isn't the only problem with the Elegoo definition files.