Ultimaker/Cura #17319: Slicing failed unexpected Error

LValadez13, 2023-11-15:
### Cura Version

5.5.0

### Operating System

Windows 10

### Printer

Ender 3 V2

### Reproduction steps

Sliced a project I printed numerous times before with no issues and this time everything I send to the slicer fails. Rebooted PC and printer, no change.

### Actual results

Nothing, Slicer still fails

### Expected results

Sliced and printed

### Add your .zip and screenshots here ⬇️

[cura.log](https://github.com/Ultimaker/Cura/files/13359855/cura.log)

GregValiant, 2023-11-15:
Thanks for the report.
_"Backend exited abnormally with return code 3221225620!"_
@rburema this looks like a blast from the past.

rburema, 2023-11-15:
Hm, except that code isn't quite the same. In hex, it's C0000094, which is the return-code for integer division by zero (as opposed to a more generic one).

Despite _everything_ failing, I think I'd still help us in this case if we had a project file from your side.

Is this the firs time slicing those projects with 5.5, or did it go well before?

GregValiant, 2023-11-15:
Doing a search I came across #8136 where this current code is mentioned down the page.

GregValiant, 2026-10-01:
I'll close this as stale.

closed 2026-10-01 by GregValiant