Ultimaker/Cura #17321: Outer Wall Wipe Distance wipes in the wrong direction

philippeholthuizen, 2023-11-15:
### Cura Version

5.5.0

### Operating System

Windows 10

### Printer

FLSUN V400

### Reproduction steps

Just as mentioned, the wipe is present, but goes in the reverse direction, thereby worsening the seam. It works ok in Cura 5.0.0 and didn't work at all in Cura 5.4.0

### Actual results

The seam is made worse by the hot end pulling the filament away from the seam location

### Expected results

The hot end should travel past the seam, thereby smoothing the seam

### Add your .zip and screenshots here ⬇️

Don't have

Khronos666, 2023-11-15:
"Outer Wall Wipe Distance" has not been working correctly since version 5.0.0.
in recent versions, it does not work if the wall is in one line, if in two, then it moves in the opposite direction...and if 3, then it works correctly...
I created a post for version 5.3.0... but it was never fixed...

->  https://github.com/Ultimaker/Cura/issues/15820

MariMakes, 2023-11-16:
Hey @philippeholthuizen,

Welcome to the UltiMaker Cura Github 🚀 
Thanks for your report 👍

We have a known issue, the one that @Khronos666 mentioned, where the wipe direction goes wrong in single/dual-walled objects. Is that the case for you as well? 

Do you have a project file for us? It contains the printer and settings we need for troubleshooting. 
To save a project file go to File -> Save project. Please make sure to .zip your project file. For big files, you may need to use WeTransfer or similar file-sharing sites.

You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser.

philippeholthuizen, 2023-11-17:
Thank you Mariska!

Yes, though it seems that the issue is slightly different in Cura 5.4 (no
wipe) and 5.5 (reverse direction). I will just use Cura 5.0 for now, where
the wipe direction is still correct.

I'll upload a project file using Cura 5.5 showing the problem this weekend

Cheers,
Philippe

emanuellonnberg, 2024-05-31:
This is still not fixed in 5.7.1. I print with one perimeter using a large nozzle and wipe would help a lot however it always goes in the wrong direction I have realized after studying my seams. Using wipe now really messes up my seams if the seam is aligned vertically as the seam gets wider and wider since not connecting the perimeter makes the next layer having nothing to adhere to and so on for each layer. I added a cylinder to the attached project that  show it as well as a an mp4 showing how it looks in preview.

https://github.com/Ultimaker/Cura/assets/2462163/791a9310-6217-4e47-9f0a-b9863dacf426

[wipe_bug.zip](https://github.com/user-attachments/files/15509085/wipe_bug.zip)
.

timhaagel, 2024-06-01:
I also encountered the issue today after noticing that my objects suddenly have "bumps" on the outer surface. I found the solution in a 2-year-old post on Reddit: https://www.reddit.com/r/Cura/comments/vwinxp/outside_wall_print_direction/. I had previously activated the "Alternate Wall Directions" function under "Experimental". Before that, I also attempted to set the number of walls from 2 to 3 and to print the outer wall before the inner wall. However, none of that helped. After deselecting the "Alternate Wall Directions" function, the simulation of the "Outer Wall Wipe Distance" seems to work correctly regardless of the number of outer walls and the printing order of the inner and outer walls.

emanuellonnberg, 2024-06-01:
Thanks! 

I checked and I had alternate walls unchecked already. Interestingly enough when I activated it the wipe did go in the correct direction on every other layer (as in the alternating layer)! but still the wrong on every other layer :(

See this video where I show first the correct wipte layer and then the next incorrect wipe layer.

Uploading wipe_correct_altearntive_dir.mp4…

emanuellonnberg, 2024-06-02:
After reading about and testing "Alternate Wall Directions" for the wipe direction I tested some more.

Turnout the "wall ordering" decides in which direction the perimeter is drawn, I previously used "inside to outside" but when I changed to "outside to inside" the wall direction was changed to clockwise and then the wipe also went in the right clockwise direction.

My guess is the bug is that the wipe always goes in the "clockwise" direction no matter in which direction the wall was drawn, so a workaround is to ensure the **outmost wall** is always drawn in **clockwise** direction.

GregValiant, 2026-10-01:
I'll close this as stale.

closed 2026-10-01 by GregValiant