Ultimaker/Cura #17322: conan helmet

dokuoireru3, 2023-11-15:
### Cura Version

5.4.0

### Operating System

windows 10 pro

### Printer

elegoo neptune 3 max

### Name abnormal settings

no

### Describe model location

no

### Describe your model

non of the above

### Add your .zip here ⬇️

the file is too large but wetransfer requires a email to send the file to. what email do i send the file to>? thank you

MariMakes, 2023-11-16:
Welcome to the UltiMaker Cura Github 🚀 
Thanks for your report 👍

If the file is too large for WeTransfer I fear that the model might not be fit for 3D printing. 
Do you at least have a screenshot for us so we can judge the geometry of the model? 

To get unstuck you can try a mesh fixer like https://myminifactory.github.io/Fast-Quadric-Mesh-Simplification/, or https://www.nanodlp.com/repair-3d-stl-file/.

I hope this helps 😇

dokuoireru3, 2023-11-16:
It isn’t too large for wefransfer, we transfer asked for a email to send the file but your form didn’t provide one I have forwarded it to this email. I can screen shot the link if need be. Thank you

GregValiant, 2026-10-01:
I'll close this as stale.

closed 2026-10-01 by GregValiant