Ultimaker/Cura #17323: Machine name not configurable on saving gcode.

nhc155, 2023-11-15:
### Is your feature request related to a problem?

Machine name not configurable on saving gcode.

I have named my printer in Cura 'MAX v3' and have configured Cura to add the machine name to the gcode file.

Cura sees my printer as 'CFFFP', and there seems to be no way to configure this.

Mildly annoying, not terribly important, but useful when you have multiple printers.

### Describe the solution you'd like

Allow the machine name to be configurable, as the printer name is.

### Describe alternatives you've considered

Ignoring it.

### Affected users and/or printers

Any printer.

GregValiant, 2023-11-15:
In the bottom left of the Cura workspace is an editable box that contains the "project name".  You can make manual changes there.
It looks to me like Cura uses a truncated version of the "definition file name" as the default prefix rather than the "machine_name" variable which is user configurable.
You are getting "CFFFP" so that would seem to indicate that you installed your printer in Cura as a "**_C_**ustom **_FFF_** **_P_**rinter" .
In my case the filename prefix is "CE3PRO" even though my machine is named "Greg's Ender 3 Pro".  I used the "creality_ender_3pro" definition file.

Right now you can manually change the name in that text box.  Somewhere internally Cura is keeping track of the definition file that was used.  I will assume that is required information when opening a project file so Cura knows the printer to install if there isn't already one of that model present.

nhc155, 2023-11-15:
Thanks, I knew about the editable box, but it would be useful not to have to remember to change the name manually every time. I do a lot of iterations when trying out a new model.
Regards,Neil H. Currie

fieldOfView, 2023-11-15:
The Print Job Naming plugin, available for free from the Marketplace,  will let you configure the prefix to anything you want.

frostyfriday, 2023-11-16:
Wow, I named my printer "Artillery Sidewinder X1" under Preferences/Printers
When saving it abbreviates it to "ASX1"
This should be documented somewhere easy to find. I have been searching for a way to change this for hours!
I was trying to find a config file somewhere, where I had changed the name to "ASX1" at some stage! Of course I couldn't find it!

Thanks for posting this. I thought I was going nuts!

I think this should be a setting because I want my file to have the prefix "x1" but naming my printer would be much nicer as it was, "Artillery Sidewinder X1"

I too eventually stumbled onto the "Custom Printjob Naming" plugin.
It is nice, but I want the print time displayed as 0305h which I can't do with this plugin as it drops the leading zero giving me 03h5 - not so nice. :(
Still better than typing it myself each time though.

nhc155, 2023-11-16:
THANK YOU !
Just what I wanted.
Cheers.

closed 2023-11-16 by nallath